firmata/ConfigurableFirmata #1: need a concept to handle Configurable Features that depend on third-party libraries

soundanalogous, 2015-04-19:
_From @ntruchsess on May 7, 2014 7:47_

Current Configurable-features all use libraries that either come with the Arduino-IDE or include their own or forked version in directory 'utility'. Not including such library into firmata utility or installing it into the Arduino library-directory would result in compile-errors even when the feature being dependent on this lib would not be enabled in ConfigurableFirmata.ino as the Arduino-ide just tries to compile everything in utility-directory.

This problem arises with #131 which depends on RCSwitch-library. We need a new way to enable/disable Configurable-features that omits compilation of Features that are not enabled. A possible solution would to move the configuration of features to a separate header-file that is included both by ConfigurableFirmata.ino and all feature-classes.

Any other thoughts?

_Copied from original issue: firmata/arduino#132_

soundanalogous, 2015-04-19:
I don't think I fully understood this issue and #150 until now (as I've been revisiting configurable firmata now that Firmata 2.4 is out). I initially thought it was an issue regarding includes, but I see now that it goes far beyond that and is a long standing issue with the way Arduino compiles libraries. Basically once you include <Firmata.h>, the IDE attempts to compile all files within the Firmata library directory and any core libraries that are dependencies of any libraries within Firmata must be included in the sketch (.ino) file.

So there are a couple of issue here:

1. Core library dependencies in Firmata feature classes (Wire and Servo are the current cases)
2. 3rd party library dependencies in Firmata feature classes (currently dependencies are included in utility)

For 1 above, including Wire.h and Servo.h in the sketch (.ino) file has sufficed, but this will not scale well if additional libraries that depend on core libraries are added.

For 2 above, we've been able to include 3rd party libraries (usually because they are modified or custom for Firmata like the stepper library) but this will obviously not scale well and there will be conflicts when people already have a version of a required 3rd party library installed.

One solution is to create use a make file, but that's not super user-friendly, especially for many Firmata users.

Another solution is to require device-specific Firmata features to be individual libraries (assuming this is possible... haven't put much thought into it yet). This may actually be more maintenance friendly since library authors will be responsible for them rather than core Firmata authors. The library simply needs to implement the proper interface to be compatible with configurable firmata. If anyone abandons a library we're not responsible for maintaining it either.

soundanalogous, 2015-04-19:
@ntruchsess, FYI. I did a quick test (using Arduino 1.5.8). I removed DHT11Firmata from the examples directory and made it a stand alone library. I was then able to include the DHT11Firmata library in ConfigurableFirmata and compile it without issue. I don't have a DHT11 sensor to verify with. This direction looks promising so far. It will help Configurable scale without bogging down the core Firmata repository. I think this direction works best for devices. Standard functionality such as I2C, OneWire, Analog & Digital I/O, etc would remain in the core Firmata repository.

soundanalogous, 2015-04-19:
Per my DHT11Firmata library experiment, I think we should require any proposals for features that wrap a 3rd party library to be implemented as stand alone Arduino library. Each library should have at least one example that shows how to use this feature in isolation (like the DHT11Firmata example does). It may optionally contain a second example integrating the feature with existing features... so it may have the new feature + all of the core ConfigurableFirmata offers). Unit tests using the [ArduinoUnit](https://github.com/mmurdoch/arduinounit) would also be helpful to ensure the interface is properly implemented (it would be up to the author as to whether or not all device-specific functionality is tested... minimally just the configurable firmata interface must be tested). A template would be provided for the tests.

The next question is where to host these libraries? These options come to mind:

1. I could create a new repo `firmata/device-libraries` or something like that. It would provide a way to do code reviews and ensure the `configurable` interface is implemented properly. This will also help ensure that a Command ID is properly allocated for each new device/feature. The main issue I have with this direction is it could create a repo with many libraries any particular user doesn't need.
2. Create an organization level account (firmata-devices) that aggregates individual libraries. This would create a single location for all libraries but allow users to clone or download individual libraries.
3. Library authors host their own libraries. A list of available libraries would be added to the configurable firmata readme and/or wiki. There would also need to be be a process of claiming a Command ID and getting that added to the core Firmata library.

soundanalogous, 2015-04-19:
I've added a proof of concept implementation of a Firmata device library here: https://github.com/soundanalogous/DHT11Firmata. If someone has a DHT11 sensor, please test. It compiles fine but I also want to verify that it actually works.

This library represents how we could handle device-specific firmata features outside of the main firmata repo.

soundanalogous, 2015-04-19:
_From @rwaldron on December 29, 2014 14:41_

Wow, thanks for deep dive on this :)

soundanalogous, 2015-04-19:
_From @MrYsLab on December 29, 2014 23:23_

I am so confused (my normal state these days). For the HC-SR04, [the j5 wiki](https://github.com/rwaldron/johnny-five/wiki/Ping#setup) has a FIrmata patch to support PulseIn. Going forward, is this the preferred method of supporting for the HC-SR04 over using NewPing for j5? If PulseIn is the way to go, will the patch be incorporated into StandardFirmata or configurable in the future? 

For Xi, my current distro, I point to the j5 wiki patch for those that want HC-SR04 support. For an older distro (PyMata) I included a one-off version of StandardFirmata that incorporated NewPing. I am most likely not going to update this latter version, and so I have no burning need for NewPing.  So should I close issue #128 or leave it open?

soundanalogous, 2015-04-19:
@MrYsLab you can close #128 if you do not plan to support NewPing. The PulseIn patch in the j5 wiki will not be added to StandardFirmata because it contains long blocking delays. Once someone creates a non-blocking pulse in/out library that supports all Arduino boards then it can be added to Firmata. However with the Firmata device library approach I'm describing above I could be less strict about whether or not a library contains blocking code or not since the support and maintenance of such libraries would fall on the device library developer rather than the core Firmata developers. It's sorta like a 3rd party plugin system for Firmata.

MrYsLab, 2015-04-20:
I went to close 128 and did not find it listed, open or closed. So if it is still open please feel free to close it.

damellis, 2015-04-20:
I like this idea; allowing people to build Firmata on top of third-party libraries seems important.

I'd suggest implementing this in a way that leaves the existing third-party libraries in their own repositories / libraries. Right now, it looks like you're including the library source-code in the Firmata repository and library, or the library-specific Firmata repository (e.g. OneWire.h and OneWire.cpp in this repository and dht11.h and dht11.cpp in the DHT11Firmata repository). This seems like it's going to make it harder to incorporate changes to those libraries. What do you think about separating the third-party libraries back out?

It seems less important to me whether or not the Firmata wrappers for individual libraries go in a single repository or in one repository per library being wrapped. Ideally, these are pretty simple files that don't have to change much, right?

soundanalogous, 2015-04-20:
@damellis that's the general idea. I believe there was a minor change to OneWire and why it was included. I'm working on a separate project, [firmata-builder](https://github.com/firmata/firmata-builder) which will provide links to download any dependencies for the collection of firmata features a user has selected.

damellis, 2015-04-20:
Sounds good!

soundanalogous, 2015-04-21:
I forgot about this, but the issue in removing the included 3rd party libraries is that a sketch won't compile unless the removed library is included, even if it isn't needed. For example, if I remove the 3rd party Encoder library, then I have to include Encoder.h in ConfigurableFirmata.ino even if I don't need it. This is because the IDE tried to compile all of the classes within the same directory as the ConfigurableFirmata.h library. I assume this is part simplicity and part time saving to avoid traversing the entire arduino library directory. The only way I've been able to get around this is to remove any wrapper libraries from the ConfigurableFirmata repo, then when you want to use them you need to include them and everything works as expected. You just have more libraries to deal with.

damellis, 2015-04-21:
Another possibility might be to put the Firmata library wrappers entirely in header files. Then they should only be compiled if they're included; if you only need to include them from the sketch, this might prevent them from being compiled unless they're used.

soundanalogous, 2015-04-22:
All code in header files works. Kind of an ugly hack but it does work. That could also fix the issue of always having to include Wire.h and Servo.h as well.

soundanalogous, 2015-04-26:
All 3rd party libraries except OneWire (which was modified) have been removed. I've also broken out what was called "EncoderFirmata" to a new library (now called "FirmataEncoder"). This new library serves as an example for creating Firmata features that wrap existing 3rd party libraries. The wrapped 3rd party libraries should not be included in the wrapper library (FirmataEncoder does not include the Encoder library). They must be installed separately.

I've also moved the I2CFirmata and ServoFirmata implementations to their respective header files. This hack fixes the issue where Wire.h and Servo.h had to be included in any sketch that included ConfigurableFirmata.h.

Next steps are to create a wrapper template (FirmataWrapperTemplate) and instructions for creating new wrapper libraries. Each wrapper library readme should list which know board architectures are supported (ideally as many as possible).

One unresolved issue is how to structure sysex commands for new wrapped features. The existing method will only scale to another 80 or so items (0x0F - 0x60 range). This may be sufficient, or a more scalable approach would be to create categories and subcategories. For example instead of a dedicated command byte for a DHT11 sensor, you'd have a TEMP_SENSOR command and then a DHT11 sub command byte.

soundanalogous, 2016-12-24:
The solution here is to externally host the Firmata feature plugin like the [FirmataEncoder](https://github.com/firmata/FirmataEncoder) example.

closed 2016-12-24 by soundanalogous