firmata/ConfigurableFirmata #15: Consider separating the concept of pin modes and configuration

soundanalogous, 2015-04-26:
Currently pin modes contain a mix of actual pin modes (digital in, digital out, pwm, etc) and configuration (Servo, OneWire, Encoder, etc). A better approach would be to separate these concepts. Pin modes should be the actual HW modes supported by each pin on a board. For example:

```
OUTPUT
INPUT
ANALOG
PWM

Instead of "I2C" the modes should be:
SDA
SCL

Instead of "SPI" (if ever added to Firmata) the modes should be:
MISO
MOSI
SCK
SS

If UART support is added the modes should be:
UART_RX0
UART_TX0
If multiple UARTs
UART_RX1
UART_TX1
etc...
```

This concept should be distinct from hardware abstractions associated with specific pins:

Servo
Shift
OneWire (unless such a thing as HW OneWire exists on a particular board)
Encoder
Stepper

damellis, 2015-04-26:
Can you say a bit more about what you're thinking? The modes related to higher-level protocols seem like they wouldn't necessarily make sense as part of a pin-mode command -- but they'd probably be useful as a way of getting the capabilities of a pin, or its current mode from the board. For example, for I2C, I imagine that you'd enable I2C as a whole, rather than individually setting one pin to SDA mode and another pin to SCL mode. (But I can imagine that it would be useful for the host to be able to find out which pin is SCL and which pin is SDA.)

Another option would be to associate the information about the mapping between pins and higher-level protocols with the protocols. That is, have a I2C message (I2C capability query or something) that would tell you which pin was SDA and which pin was SCL. In that case, you might actually want to keep the pin mode as I2C, not SDA/SCL. That is, it might be good if the core messages (like pin-mode) don't need to know any details about the individual high-level protocols (like their individual pins); they'd just need to know about the existence of the protocols. That is, it might be simpler if the pin-mode messages can just say something like "this pin is used by the I2C protocol" and then the details are all in I2C-specific messages. It just seems like it might separate the abstractions better. Just a thought.

Also, some microcontrollers might have more than I2C bus and more than one SPI bus.

soundanalogous, 2015-04-26:
I copied this issue over from the firmata/arduino repo so it's a bit outdated. My latest thinking here is to use the resolution byte of the capabilities response for the pin constant. So for I2C the capability response would return something like this for pin that has I2C functionality. For example, [this block of code](https://github.com/firmata/ConfigurableFirmata/blob/master/src/I2CFirmata.h#L148-L154) could be updated to:

```c++
void I2CFirmata::handleCapability(byte pin)
{
  if (IS_PIN_I2C(pin)) {
    Firmata.write(I2C);
    Firmata.write(pin == SDA ? 0 : 1); // 0 = SDA, 1 = SCL
  }
}
```
or

```c++
// removes need for IS_PIN_I2C macro in Boards.h
void I2CFirmata::handleCapability(byte pin)
{
  if (pin == SDA || pin == SCL) {
    Firmata.write(I2C);
    Firmata.write(pin == SDA ? 0 : 1); // 0 = SDA, 1 = SCL
  }
}
```

Kind of a hack but the resolution as it is currently implemented only has meaning for Analog and PWM (resolution in bits) so it could be used to communicated specific pin types for non-analog features. It would also likely maintain backwards compatibility since the resolution value was not likely used previously (aside from Analog and PWM which would remain unchanged).

The other thing I need to do is create a more clear distinction between pin capability and pin mode. I see this has been confused in ConfigurableFirmata in some places (probably my fault). This is illustrated [here](https://github.com/firmata/ConfigurableFirmata/blob/master/src/I2CFirmata.h#L290-L291). `setPinMode` in Line 290 should really renamed to be `setPinConfig` since [that is actually what is being stored](https://github.com/firmata/ConfigurableFirmata/blob/master/src/ConfigurableFirmata.cpp#L445). Line 291 is a bug, `pinMode` should not be called at all since it's handled internally in the Wire library.

soundanalogous, 2015-05-03:
The other thing I'm struggling to make a decision on here is whether or not certain features even need to respond to a configuration query. I think features related to the capability of the microcontroller should be reported (ANALOG, INPUT, OUTPUT, PWM, I2C, SPI, etc). However I'm not sure that device (or device type) features need to be reported. Servo had to be reported because of the way it used the analog message to pass the value. However I don't think it makes much sense for other device-specific features to return anything for a configuration query since I see the configuration query as providing value in reporting the configuration of the microcontroller pins rather than the peripherals attached to the microcontroller.

Other reasons are that any feature is included in a configuration query response needs a unique "pin mode". Often these unique "pin modes" also need macros in Boards.h (IS_PIN_I2C, IS_PIN_SERVO, etc). I really don't want to have to add a new pin mode and set of macros for each type of device supported, especially when most of those devices could be added to any digital pin on the board (or perhaps in some cases, any analog pin). I was also able to eliminate the need for IS_PIN_SERVO and PIN_TO_SERVO macros in Boards.h as of Firmata v2.4.0 (the macros are still included in Boards.h but are no longer used in StandardFirmata or ServoFirmata).

If the configuration query is specific to the microcontroller, then the pin state query should be specific to the microntroller as well. It should return the state value for the HW function of the queried pin. This avoids the mess of worrying about state values for specific devices (like the state of DHT11 temp/humidity sensor). The state of a device should be handled by the protocol for that device (perhaps an interface to querying the state of a device can be defined).

So the decision I'm looking to make here is if it's save to say configuration query and pin state query should be specific to the functionality of the microcontroller going forward. The SERVO pin type will be deprecated once the new sysex-only servo implementation is added (avoids the weird use of analog message). the ENCODER pin type will be removed and an interface will be defined for querying devices for their configuration.

Thoughts?

damellis, 2015-05-04:
If a given feature only works on certain pins, it seems like you need a way to find out which pins those are (e.g. older versions of the Servo library only worked on pins 9 and 10 because of their use of a particular AVR timer). But I think you're right that if a feature works on any pin, you probably don't need to include it in the capability list of pins.

On the other hand, even if a feature works on any pin, you may want a way to specify or discover which feature is using a given pin. For example, if I reconnect to a board that's already running Firmata (without resetting the board), I may want to discover which pins are being used to generate servo pulses. I would think you'd want to do this by reporting the ID of the feature itself (i.e. the same ID used to send sysex messages for that feature) rather than by having a parallel pin configuration ID corresponding to each feature. (You'd probably just need a single pin configuration ID that means "used by a sysex feature" and then a way to query the sysex feature ID for each pin.)

soundanalogous, 2016-12-24:
This is being addressed in the protocol documentation: https://github.com/firmata/protocol/issues/41

closed 2016-12-24 by soundanalogous