firmata/ConfigurableFirmata #18: Is there a reason "set digital pin value", 0xF5 from protocol spec is not implemented?

markgrebe, 2015-06-10:
Set digital pin value, command 0xF5, does not seem to be implemented in either Standard or Configurable Firmata.  As this could be simply built on top of the digitalWrite function, I'm curious as to why it is omitted.  Also, with the addition of the Scheduler in Configurable Firmata, there is no way to set the value of just one digital pin from a scheduled task.  With the addition of the F5 command, it would then be possible to do the same types of digital tasks from the host or from a scheduled task.
I am writing an updated Firmata client for Haskel as part of my grad school research, and having the ability to set one digital pin would be extremely helpful 
I'm more than willing to create a proposed update if others are too busy.

soundanalogous, 2015-06-11:
It is implemented in StandardFirmata here: https://github.com/firmata/arduino/tree/v2.5.beta1. So far no one has given me any feedback so it has remained in beta until it's tested by someone other than myself. I haven't had time to add it to ConfigurableFirmata now. I only have a couple of hours/week to work on Firmata most weeks so it's very slow progress unfortunately.

markgrebe, 2015-06-11:
Thanks!  The Set Digital Pin Value works great in the Beta.  I haven't done any large testing of anything but that feature in the Beta, but I will be using it now, so I can also let you know if I find any issues.  I totally understand the slowness of part time open source efforts, and appreciate the work you do put into it.  I'll let you know if I get to adding it to the Configurable side before you do.

closed 2015-06-11 by markgrebe