firmata/ConfigurableFirmata #19: ATH0

samkuhn, 2015-06-23:
I am currently using [this](https://gist.github.com/edgarsilva/e73c15a019396d6aaef2) modified Firmata since I have node.js running directly on my (disk expanded) Yun. 

The problem is that it does not support OneWire (needed for a DS18B20 digital temperature sensor). So now I am looking into using ConfigurableFirmata instead. But I would need to get ConfigurableFirmata to work over ttyATH0 as the above Firmata does. Is this possible?

soundanalogous, 2015-06-24:
Try this: https://gist.github.com/soundanalogous/b3093c15ccd14c6b2149

soundanalogous, 2015-06-24:
It may be too large to compile. If that's the case, uninclude any unused features (such as Encoder, etc).

samkuhn, 2015-06-24:
Thanks! I actually worked it out shortly after posting (and didn't get time to reply sorry). For anyone coming to this issue the key part to getting a Firmata to work over ttyATH0 is changing:

```c
Firmata.begin(57600);
```

to:

```c
Serial1.begin(57600); 
Firmata.begin(Serial1); 
```

(as in soundanalogous's Firmata above)

closed 2015-06-24 by samkuhn