firmata/ConfigurableFirmata #21: Connection issue

PitySOFT, 2015-09-19:
Hello,

I'm trying to use johnny-five with node.js to connect over LAN with an Arduino Uno with Ethernet Shield.

If I use the StandardFirmataEthernet library, I get the blink let just fine, but with that library I can't use OneWire sensors. When I use the ConfigurableFirmata library with the same IPs configuration give's me timeout with following error:

"Device or Firmware Error A timeout occurred while connecting to the Board."

Any idea why, or what I can do to make it work?

Thanks.

soundanalogous, 2015-09-27:
Please paste the code you are uploading to the Arduino and separately your johnny-five code so I can take a look.

PitySOFT, 2015-09-27:
The code it's just the a testing code, nothing complicated. If I use same settings with the StandardFirmataEthernet the blink works.

This is what I have configured in the ConfigurableFirmata with all the comments removed.

```C#
#include "ConfigurableFirmata.h"

#include <SPI.h>
#include <Ethernet.h>

//#include <UIPEthernet.h>

//#include <Bridge.h>
//#include <YunClient.h>

#if defined ethernet_h || defined UIPETHERNET_H || defined _YUN_CLIENT_H_
/*==============================================================================
 * Network configuration for Network Firmata
 *============================================================================*/
#define NETWORK_FIRMATA
#define remote_ip IPAddress(192,168,1,53)
//#define remote_host "server.local"
#define remote_port 3030
#define local_ip IPAddress(192,168,1,40)
const byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};
#endif

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;
/*
#include <Servo.h> //wouldn't load from ServoFirmata.h in Arduino1.0.3
#include <ServoFirmata.h>
ServoFirmata servo;

#include <Wire.h> //wouldn't load from I2CFirmata.h in Arduino1.0.3
#include <I2CFirmata.h>
I2CFirmata i2c;

#include <OneWireFirmata.h>
OneWireFirmata oneWire;
/*
#include <StepperFirmata.h>
StepperFirmata stepper;

#include <FirmataExt.h>
FirmataExt firmataExt;

#include <FirmataScheduler.h>
FirmataScheduler scheduler;
*/
```

And this is the node.js code.

```Javascript
var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({
    port: new EtherPort(3030)
});

board.on("ready", function () {
    var led = new five.Led(8);
    led.blink(500);
});
```

soundanalogous, 2015-09-27:
You need to include FirmataExt or nothing in ConfigurableFirmata will work.

PitySOFT, 2015-09-27:
Thanks for the answer.

I've included that and still a timeout response, also I've tried with all the libraries and same response.

PitySOFT, 2015-09-27:
It seems that the network cable was the problem on the last message. Now I can read onewire devices! Thanks!

closed 2015-09-27 by PitySOFT