firmata/ConfigurableFirmata #22: Can not compile with teensy 3.1

taoyuan, 2015-10-16:
Sample `ConfigurableFirmata.ino` can not compile in teensy 3.1 with errors:

```
In file included from /Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/OneWireFirmata.h:17:0,
from ConfigurableFirmata.ino:110:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/utility/OneWire.h:109:2: error: #error "Please define I/O register types here"
#error "Please define I/O register types here"
^
In file included from ConfigurableFirmata.ino:110:0:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/OneWireFirmata.h:37:0: warning: "ONEWIRE_CRC" redefined [enabled by default]
#define ONEWIRE_CRC 0 //for OneWire.h: crc-functions are not used by Firmata
^
In file included from /Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/OneWireFirmata.h:17:0,
from ConfigurableFirmata.ino:110:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/utility/OneWire.h:30:0: note: this is the location of the previous definition
#define ONEWIRE_CRC 1
^
In file included from ConfigurableFirmata.ino:103:0:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/ServoFirmata.h: In member function 'boolean ServoFirmata::analogWrite(byte, int)':
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/ServoFirmata.h:70:1: warning: no return statement in function returning non-void [-Wreturn-type]
}
^
In file included from /Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/OneWireFirmata.h:17:0,
from ConfigurableFirmata.ino:110:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/utility/OneWire.h: At global scope:
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/utility/OneWire.h:116:5: error: 'IO_REG_TYPE' does not name a type
IO_REG_TYPE bitmask;
^
/Users/taoyuan/Documents/Arduino/libraries/ConfigurableFirmata/src/utility/OneWire.h:117:14: error: 'IO_REG_TYPE' does not name a type
volatile IO_REG_TYPE *baseReg;
^
```

Changed [line 64 in utility/OneWire.h](https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/OneWire.h#L64)
```
#elif defined(__MK20DX128__)
```
to
```c
#elif defined(__MK20DX128__) || defined(__MK20DX256__)
```

It can compile. Just do not know wether is correct ?

PaulStoffregen, 2015-10-16:
Yes, correct.  Just take a look at the latest code (and perhaps update your old copy)

https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h#L64

taoyuan, 2015-10-16:
@PaulStoffregen Thanks for clarifying. 

Maybe the `OneWire` of `ConfigurableFirmata` should fix this.

soundanalogous, 2015-10-16:
I'll update the copy of OneWire in ConfigurableFirmata for a short-term fix. Longer term, I'd like to use Paul's OneWire lib without any modifications, I need to resolve [this difference](http://www.beleaderly.com/ghc15/) first (It was added by the original Configurable firmata author and was the reason for the copy of OneWire included with Configurablefirmata).

soundanalogous, 2015-10-17:
@PaulStoffregen, FYI I noticed when testing this that Teensy 3.x boards won't compile if Servo library v1.1.0 is installed (installed via library manager - required for Arduino Zero). It compiles fine when I fall back to Servo v1.0.3 (which is the version included with Arduino 1.6.5).

This is the error message I get when compiling for Teensy 3.1/3.2 or Teensy 3.0:
```
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp: At global scope:
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:80:8: error: expected constructor, destructor, or type conversion before '(' token
SIGNAL (TIMER1_COMPA_vect)
^
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp: In function 'void initISR(timer16_Sequence_t)':
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:128:5: error: 'TCCR1A' was not declared in this scope
TCCR1A = 0;             // normal counting mode
^
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:129:5: error: 'TCCR1B' was not declared in this scope
TCCR1B = _BV(CS11);     // set prescaler of 8
^
In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/wiring.h:33:0,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/WProgram.h:15,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/Arduino.h:1,
from /Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:23:
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:129:18: error: 'CS11' was not declared in this scope
TCCR1B = _BV(CS11);     // set prescaler of 8
^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/core_pins.h:45:22: note: in definition of macro '_BV'
#define _BV(n)  (1<<(n))
^
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:130:5: error: 'TCNT1' was not declared in this scope
TCNT1 = 0;              // clear the timer count
^
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:136:5: error: 'TIFR1' was not declared in this scope
TIFR1 |= _BV(OCF1A);     // clear any pending interrupts;
^
In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/wiring.h:33:0,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/WProgram.h:15,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/Arduino.h:1,
from /Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:23:
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:136:18: error: 'OCF1A' was not declared in this scope
TIFR1 |= _BV(OCF1A);     // clear any pending interrupts;
^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/core_pins.h:45:22: note: in definition of macro '_BV'
#define _BV(n)  (1<<(n))
^
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:137:5: error: 'TIMSK1' was not declared in this scope
TIMSK1 |=  _BV(OCIE1A) ; // enable the output compare interrupt
^
In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/wiring.h:33:0,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/WProgram.h:15,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/Arduino.h:1,
from /Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:23:
/Users/soundanalogous/Documents/Arduino/libraries/Servo/src/avr/Servo.cpp:137:20: error: 'OCIE1A' was not declared in this scope
TIMSK1 |=  _BV(OCIE1A) ; // enable the output compare interrupt
^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/core_pins.h:45:22: note: in definition of macro '_BV'
#define _BV(n)  (1<<(n))
```

closed 2015-10-17 by soundanalogous

PaulStoffregen, 2015-10-18:
The problem is copy of Servo which you manually installed is overriding the built-in copy that comes with Teensyduino.

soundanalogous, 2015-10-18:
So that would be an issue with the Arduino Library manager.