firmata/ConfigurableFirmata #27: Timer-driven stepper implementation

soundanalogous, 2015-10-18:
_From @soundanalogous on June 4, 2013 2:16_

Create a timer based Stepper motor implementation for ConfigurableFirmata. The current Stepper implementation does not use a timer since all were already in use by StandardFirmata (at least for an Arduino Uno and similar boards). With ConfigurableFirmata a timer-based Stepper implementation should be possible as long as the user does not enable another feature that uses the same timer.

_Copied from original issue: firmata/arduino#60_

DomAmato, 2015-10-20:
Since the Servo and Tone libraries are tied to certain timers how should we handle cases where a servo or tone is added after the stepper? are the "native" timer implementations the ones that override or is it a first come first served thing, though an error message of some kind should probably be sent.

soundanalogous, 2015-10-20:
Will probably need a scheme for throwing compiler errors when users attempt to mix incompatible features for a given architecture.

PaulStoffregen, 2015-10-20:
Please keep in mind non-AVR boards have very different timer resources.

For example, on Teensy 3.1 & 3.2, Servo uses the "PDB" timer and tone uses one of 4 "PIT" timers, so there's no conflict, *and* neither of those are used for PWM generation.  Teensy LC doesn't have the PDB and it only has two PITs, but there Servo uses "LPTMR", so it can still implement Servo and tone and leave one PIT timer available for other libraries.  But even on Teensy 3.2 with 12 different hardware timers of various types, some combinations of timer-based features and libraries do conflict.

DomAmato, 2015-10-20:
Sounds like it might be easier then to create a separate timer module that would handle timer assignment. I think it would be a lot of extra work to handle every timer case on a per module basis. The helper class would essentially find the board type and keep track of timer assignments based on what modules have been requested. It can then send an error when libraries conflict or if more timers have been requested than currently exist

soundanalogous, 2015-10-20:
That would be interesting to explore. This feature is low priority for me but if you or anyone else wants to take it on it could happen sooner than later.

DomAmato, 2015-10-20:
I can put something together this week, ill look up what I can about currently supported boards. I might just put together a proposal of what it would look like implementing just the atmega328 so that other boards can be added using a similar style

soundanalogous, 2015-10-20:
It may be better to avoid the ATMega328 (and similar) boards for new and updated timer-related features and only enable it for Teensy3.x and SAM-based boards - those with more timers. Otherwise there are too many potential error cases and users are likely to be more frustrated than anything else. Most users who turn to Firmata are looking for a one-size-fits-all solution where they can install software once (ie StandardFirmata). Obviously there are limitations to this. With the newer, more capable boards we can stretch the possibilities a bit, but with the less capable boards we are already very near the limit with StandardFirmata. I was hoping configurable firmata would be a way to address these shortcomings, but based on feedback I've received so far many users are not interested in the piecemeal approach.

It would be better in many cases for users looking for more functionality (especially when it involved higher precision) to write their own software for the Arduino board and then craft a simple protocol (or use an existing protocol) to expose an appropriate interface for their application.

soundanalogous, 2015-10-20:
Also, I filed this issue a couple of years ago on firmata/arduino and moved it over here last weekend as I was moving all of the configurable firmata specific stuff out of firmata/arduino and into firmata/ConfigurableFirmata

DomAmato, 2015-10-21:
What about the mega? it has 6 timers which is about half the teensy 3s. I can see the reason for the 328 since it only has 3 same with the teensy 2 and the 2++ only has 4, the SAM boards have 9? I guess what I am getting at is should we be setting a minimum hardware spec in terms of timers if we are going to make a helper class? I think the megas can still be useful with what is essentially 5 assignable timers

soundanalogous, 2015-10-21:
Yeah we can consider Mega as well. However I think if you are also using Servo, additional timers are used to allocate more Servos. I'd need to look it up to be sure, but I think Mega uses up to 3 timers for Servo whereas Uno uses 1 timer.

soundanalogous, 2015-10-21:
Another thing to think about here is how bad is the current stepper implementation vs what a timer-based implementation would be considering time and effort, etc. You probably have a pretty good idea here based on all the work you've done in your branch on steppers.

PaulStoffregen, 2015-10-21:
This may or many not help, but from feedback I've heard, the AccelStepper library is the most popular timer-based non-blocking library.  Maybe it's worth considering?

soundanalogous, 2015-10-21:
AccelStepper is nice. It's non-blocking but not timer-based, it simply checks the time on each call to run in the main loop. It would probably be better to use than the [similar non-blocking stepper lib I hacked together (https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/FirmataStepper.cpp) for Firmata. There was some reason I opted not to use it when I first added stepper support but I can never remember what that exact reason was.

DomAmato, 2015-10-22:
It should be easy enough to attach the update to a timer though, which is sort of a halfway solution but I know that library is pretty well written. certainly a good starting point

PaulStoffregen, 2015-10-22:
The TimerOne library might be useful?  It provides an abstraction for the AVR timer.  It also has been ported to the 32 bit Teensy boards.  Maybe it could eventually also be ported to Arduino Zero & Due, and other boards like ESP8266?  Long-term, the Arduino world really needs some dependable hardware independent timer abstractions....

soundanalogous, 2015-10-22:
That is an interesting idea and would keep the timer abstraction decoupled from the stepper library.

soundanalogous, 2015-11-29:
Now that ConfigurableFirmata is much easier to use due to http://firmatabuilder.com, I've decided wrapping AccelStepper and rethinking the Firmata stepper interface is the best approach. I've created a new issue to track the work: https://github.com/firmata/ConfigurableFirmata/issues/33 and will close this issue.

closed 2015-11-29 by soundanalogous