firmata/ConfigurableFirmata #28: Add Arduino pluggable usb midi to ConfigurableFirmata

EchoEOL, 2015-11-08:
Could support for Arduino 1.6.6 pluggable usb midi for ConfigurableFirmata be added as this could allow Firmata to be used directly with Remote Scripts for Ableton Live directly or passed straight into M4L without any further external bridges?

soundanalogous, 2015-11-09:
So you want the only communicate with the Arduino from Ableton Live over usb midi instead of serial? If this is the case it is probably possible but may require a new project since usb midi doesn't implement the Stream interface that Firmata uses for the Arduino implementation. A new usb midi specific version of Firmata  (FirmataUSBMidi or something like that) would need to be created.

soundanalogous, 2015-11-09:
However if what you are looking for is a separate control line for usb midi, that may be possible with ConfigurableFirmata. It would require a serial connection the computer to send Firmata commands and a separate usb midi connection to the computer to communicate with Ableton. Serial communication would go through Serial1 (or Serial2 or Serial3 depending on the board).

EchoEOL, 2015-11-14:
I was thinking about direct (USB) MIDI communication. 

I did look at pluggable usb midi and ConfigurableFirmata.myself but I'm new to C++ and got lost pretty fast. I had mistaken the TCP/IP stream as a possible quick fix???

My initial thinking was to have a go a patching Firmata to return note on / note off for the digital input pins and MIDI's 14-bit control messages (CC 32-63 are matched to CC 0-31) so that I could send 14 bit messages back from the ADC outputs. I know that this would take the protocol outside of SYSEX but a Freescale MPXV4006GP pressure sensor added to an ADC input would give me a USB MIDI breath controller that could be used by any music software with MIDI input for example. While device configuration still uses SYSEX.

I now realise that this is now an almost totally different project but I initially thought that I could break the back of the project using Firmata

soundanalogous, 2015-11-14:
It may be possible to create a Stream wrapper for USB Midi (creating a new class that bridges USB midi's methods to the Stream base class methods). Then USB Midi could work with ConfigurableFirmata as currently implemented. I don't know why Arduino didn't extend Stream in the first place, the public interface is so close.

soundanalogous, 2015-11-14:
Public interface of MIDIUSB class: https://github.com/arduino-libraries/MIDIUSB/blob/master/src/MIDIUSB.h#L213-L219

Public interface of Stream base class: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.h#L48-L53

Public interface of Print (Stream extends Print and adds the `write` method):
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Print.h#L43-L81

Example of a Class that implements the Stream interface: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.h

See how Hardware Serial uses the methods defined by Stream (and `write` from Print). You'd create a new class that extends Stream and implements these methods (`available`, `read`, `flush`, `write`, `peek`) and those methods would call the similar methods of the MIDIUSB class. You'd leave peek as empty since there is no equivalent in MIDIUSB. You'd have to figure out what to do with `sendMidi` since it doesn't have a match in Stream. You'd also have to add a `begin` method to the wrapper class since this is necessary. I'm not sure what this needs to do with MIDIUSB - you'd essentially do any required initialization. You then include the wrapper class in your Arduino Sketch and pass this into the Firmata.begin method like this:

```c++
// in your ino file (a clone of ConfigurableFirmata.ino, with the ethernet code stripped out).
MIDIUSBWrapper stream(/*required params*/);

Firmata.begin(stream);
```