firmata/ConfigurableFirmata #37: serial receive prevents reconnect to host

jnsbyr, 2015-12-31:
- Hardware: Teensy LC with ENC28J60 LAN module
- Software: Configurable Firmata 2.8.0, UIPEthernet 1.09
- Active Features: digital outputs, I2C, serial port

Operation is as expected until a LAN disconnect occurs and serial data is received while disconnected. 

Firmata will not attempt a reconnect in this case and must be cold booted. If no serial data is received while disconnected, a successful TCP reconnect will be done by the same firmware build.

The serial data gets directly forwarded to the UIPEthernet driver - typically one byte per telegram, because the Teensy seems to be fast enough for this. Each received byte is multiplied by 6 because of the telegram overhead. At 9600 baud this can amount to 30.000 bytes in 5 seconds. Probably an overflow occurs somewhere before the disconnect can be detected and further write operations can be skipped by checking EthernetClientStream::maintain() in EthernetClientStream::write(). It is desirable to prevent the flooding of the Firmata transmitter, maybe by checking buffer sizes or return values.

soundanalogous, 2015-12-31:
When you say serial data are you talking about a data from a serial peripheral connected to the Teensy (such as a GPS module) or are you talking about a serial connection to a computer?

soundanalogous, 2015-12-31:
The author of EthernetClientStream and the Ethernet connection in general is no longer active on this project. This will be a bit difficult for me to fix so I may need some help. The first thought I have is I could try to make it so that data is only read or written while the ethernet connection is active.

Maybe moving this to the top of `loop()` would help:
```c++
if (Ethernet.maintain()) {
  stream.maintain(Ethernet.localIP());
} else {
  return; // don't read or write any data while disconnected
}
```

jnsbyr, 2016-01-01:
Happy New Year, soundanalogous!

As you already assumed, I am talking about serial data from a peripheral device. 

I the meantime I tried something similar by checking Firmata::available before adding new data for transmit in SerialFirmata::checkSerial but the result was unexpected. Serial data was no longer forwarded byte by byte but mostly discarded. But once in a while a complete telegram was transmitted. The reconnect issue was not affected by this modification.

I took your suggestion but sadly with the same result in regard to the reconnect. 

After Firmata gets disconnected it is still possible to ping the Arduino, so Firmata has not crashed. I assume that something goes wrong with the send buffers before Ethernet.maintain can detect that the connection is broken.

I already considered trying a W5100 LAN module to see if this problem is more on the LAN driver or on the Firmata side but I am not using a breadboard so rewiring is not that easy.

jnsbyr, 2016-01-02:
Changing the LAN-Module from ENC28J60 to W5100 shows that the reconnect issue seems to be caused by the UIPEthernet driver.

The Firmata reconnect works as expected even with serial receive when using the W5100 Module.

soundanalogous, 2016-01-02:
I personally never had any luck with the ENC28J60 (based on one cheap shield I purchased off eBay) which is why I left the UIPEthernet driver out of [StandardFirmataPlus](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino). I assume it works with some ENC28J60-based shields and adapters since the author of the UIPEthernet driver also added support for it to ConfigurableFirmata. You could try opening an issue for the UIPEthernet library here (if that same issue is not already open): https://github.com/ntruchsess/arduino_uip/issues.

jnsbyr, 2016-01-02:
Thanks for your support. I will contact Norbert about this.

jnsbyr, 2016-01-03:
After less than 10 hours continuous operation the setup with the W5100 also failed - this time not caused by a reconnect but in mid operation by ceasing to transmit to the host and being unresponsive to commands from the host, including reset.

netstat on the host side (RPi) showed an unclosed connection, pinging Firmata from the host was still possible. Restarting the host application did not make the Firmata reconnect, even after waiting for almost an hour.

I can only guess at possible reasons. Probably the host did not process all received telegrams within milliseconds at some time because it hat something else to do for a (few?) seconds, so there was a slight backlog, the TCP receive buffer became full on the host side and the TCP transmit buffer getting filled on the Firmata side until there was a buffer overrun.

May idea is to check the TCP transmit buffer size before queueing new telegrams to the host. Additionally a serial data receive collect delay (similar to the Nagle algorithm) may help reducing the overhead cause by single received serial byte forwarding. In my opinion the issue cannot be specific to continuous serial receive forwarding but should also arise if polling analog inputs at higher rates, but I do not have the required setup to confirm this.

Currently I am running a cross-check by disconnecting the serial line and running Firmata without serial transfer.

soundanalogous, 2016-01-05:
Here's another older issue reporting a crash after several hours: https://github.com/firmata/arduino/issues/177 although it doesn't seem related.

In your particular issue is it still related to the use of the UART or is it a general connection failure after several hours (regardless of if the UART is in use or not).

Also, regarding Serial and buffering. The Serial input is actually buffered, but the loop iteration is so fast you only see one byte transmitted at a time (especially if using 9600 baud). I considered adding command that can set an interval for when `checkSerial()` is called. The a user can fine tune how the transmission by choosing an appropriate interval along with the number of bytes to send per interval (`maxBytesToSend` - which is already available). This is a feature I could still add if necessary. I was waiting to see if I really needed it because it adds complexity.

soundanalogous, 2016-01-10:
Just had a thought... are you using DHCP? If you are I wonder if the lease is expiring after a few hours and the local IP is not being maintained appropriately. [This code](https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino#L338-L343) should make sure the local IP is reset if the lease expires. Perhaps there is an issue here (if you are using DHCP).

jnsbyr, 2016-01-10:
Thanks for the suggestion, but I wanted to avoid any issues with DHCP, so I used a fixed IP from the start. 

An interesting aspect is the unchanged traffic indication on the TX LED of the RJ-45 port after the disconnect. I intend to use tcpdump to check what is happening here. Though ENC28J60 and W5100 are behaving differently there might be something to do with the host side of the connection.

jnsbyr, 2016-01-17:
To make sure there are no side effects caused by my setup, I reinstalled/updated all software components:
* Arduino IDE 1.6.7
* Teensyduino 1.27
* up-to-date Ethernet for Teensy (https://github.com/PaulStoffregen/Ethernet)
* ConfigurableFirmata 2.8.0

My test was performed using the following hardware
* Teensy LC with W5100 LAN module
* RPi 2.0 with Rasbian 4.1 as host

Firmata features in use:
* digital outputs (not changed during test)
* I2C with 2 devices (used periodically every 5 minutes)
* serial port (almost continuous receive at 9600 baud)

Test results:
* as before operation ceased after a few hours
* netstat on RPi showed socket as still connected
* tcpdump showed that packets could still be send from host to Firmata and that these packets were ACKed by Firmata but no packets where send by Firmata to the host
* closing TCP client connection on host side did not trigger Firmata reconnect
* manual Firmata reboot was required to restore operation

In the StandardFirmata.ino and in ConfigurableFirmata.ino there is an interesting comment in the main loop:

```C
/* SEND STREAM WRITE BUFFER - TO DO: make sure that the stream buffer doesn't go over
   * 60 bytes. use a timer to sending an event character every 4 ms to
   * trigger the buffer to dump. */
```

FirmataClass seems to use single byte writes to the EthernetClientStream so there are no buffers at the top level. EthernetClientStream, Stream and Print do not have members used for buffering. So is the "send stream write buffer" referred to maybe the one in the W5100 driver? If the host is busy and cannot pickup all received bytes immediately, 60 bytes are quickly waiting to be sent on the Firmata side when having continuous serial receive. Currently I do not have the oversight to fix this problem without a hint to the probable cause from someone who has more experience with the architecture in use. Any suggestions in this respect are welcome.

My other idea for a little more reliability at this time is a workaround: monitor received telegrams from the host and perform a reset on the Firmata side if timed out. This would require that Firmata cannot receive successfully in the above described situation (neither proven nor disproven yet) but the main loop is still running. A Firmata built-in alive-telegram with a counter value that has to change (e.g. Firmata zero to host, host echo value, Firmata increment value to host, host echo value ...) would yield more reliable results. Using a COP would also be an option but this would not be a hardware independent approach.

soundanalogous, 2016-01-17:
Have you tried running this for several hours leaving the serial device out in order to isolate whether or not a potential overflow of the serial buffer is actually the problem?

soundanalogous, 2016-01-17:
Also this comment:
```
/* SEND STREAM WRITE BUFFER - TO DO: make sure that the stream buffer doesn't go over
   * 60 bytes. use a timer to sending an event character every 4 ms to
   * trigger the buffer to dump. */
```
Has been in the code for years (since Firmata only ran over a Serial transport). I have not written any buffering code for Firmata. For Serial, the Arduino Serial libraries (including SoftwareSerial) handle buffering internally. I have not idea if the Ethernet library buffers internally or not as I've never read the code. Networking is not in my core area of expertise, but I guess I'll have to learn now.

soundanalogous, 2016-01-17:
> tcpdump showed that packets could still be send from host to Firmata and that these packets were ACKed by Firmata but no packets where send by Firmata to the host

This may not indicate that Firmata is actually responding. These Ethernet modules do a lot of work on their own. I'll create a branch of ConfigurableFirmata that we can use to diagnose the issue. That code will also report the memory usage of ConfigurableFirmata over time so we can check if Firmata stops responding after a few hours due to a memory leak.

soundanalogous, 2016-01-17:
https://github.com/firmata/ConfigurableFirmata/tree/issue-37

jnsbyr, 2016-01-17:
For me this is not a breadboard experiment (some things would be easier to test but it is not easy to replicate the setup). Because of the current instability with serial port operation I disconnect the serial cable after an unsuccessful test to keep the rest running. Therefore I can say that with serial pin function enabled but without serial input the stability changes from a few hours to at least several days. 

I have another system running using a Pro Mini 5V clone, StandardFirmataEthernet 2.5 and the same W5100 LAN module with digital inputs, digital outputs and pwm outputs. This system is running for several weeks without issues.

So both systems have the same network device but different CPUs. I would prefer to use a Pro Mini for the serial port test but the Pro Mini has only one serial port and the first serial port of a system is not available to Firmata - this is why I use a Teensy LC. My initial setup was with a ENC28J60 LAN module. With the ENC28J60 stability seemed to be better but reconnecting was not possible at all (see my initial comment of this thread), while the W5100 is able to reconnect but has a stability issue. Without continuous serial receive the setups seemed to be stable.

I have already seen the branch you created to debug the issue. Is it possible to send the debug information over the host link? My Firmata setup is in a location where I would need a notebook for serial debugging. That is also the main reason I want to use a serial port with Firmata over LAN - it's the perfect COM server.

soundanalogous, 2016-01-17:
It's possible to send debug messages. It uses the Firmata `SERIAL_MESSAGE` command so as long as this is implemented in your host application you can receive debug messages. Are you using an existing Firmata client library in your host application or did you write your own?

Also, please share your configurable firmata sketch so I can add that to the debug branch.

soundanalogous, 2016-01-17:
I have not made any changes to the issue-37 branch yet. Once I get your sketch, I'll add free memory logging.

jnsbyr, 2016-01-17:
I use a modified version of the current official release of FHEM as host application (see recent merges in https://github.com/ntruchsess/fhem-mirror and https://github.com/ntruchsess/perl-firmata). If SERIAL_MESSEGE support is missing, I will make the required changes.

You can find my sketch here: [ConfigurableFirmata.zip](https://github.com/firmata/ConfigurableFirmata/files/93565/ConfigurableFirmata.zip)

soundanalogous, 2016-01-17:
SERIAL_DATA is implemented in perl-firmata: https://github.com/ntruchsess/perl-firmata/blob/master/lib/Device/Firmata/Protocol.pm#L283.

jnsbyr, 2016-01-17:
The SERIAL_DATA telegram handling in FHEM is the commit I have provided. 

In my setup the serial data will be passed to an telegram parser for a Bosch Heatronic 3 gas boiler controller. Can you please add a unique prefix+postfix to the debug message (e.g. +++DEBUG ... DEBUG---). Then I can extract it at protocol handler level.

soundanalogous, 2016-01-17:
Sorry, I meant STRING_DATA.

soundanalogous, 2016-01-17:
Turns out the memory check routine I normally use does not work for Teensy LC. Will need to do some research.

jnsbyr, 2016-01-17:
Sorry for the trouble with the Teensy - but according to Boards.h the only other Arduinos with more than one serial port are the Mega and Duo and they are just too big for my purposes.

soundanalogous, 2016-01-17:
I just pushed a [commit](https://github.com/firmata/ConfigurableFirmata/commit/75d1ed26a6308269d122962a42e4d6e8c3d8fe2e) to the issue-37 branch. This will report the free memory available on the Teensy LC every 10 minutes. The changes are made in /examples/issue37.ino so you'll want to compile and upload that file. You can also change the reporting interval - see the `memCheckInterval` on line 37 of issue37.ino.

I also moved the input stream processing before the call to `Ethernet.maintain` in the main loop. However this likely won't make any difference.

The other thing I can try to figure out is how to dump the Serial read buffer while Ethernet is disconnected. Even though serial.update() is not called when Ethernet is disconnected,, the buffer is still filling up as long as the Serial peripheral is attempting to transmit data.

soundanalogous, 2016-01-17:
Another update to discard serial data from the peripheral while Ethernet is disconnected: https://github.com/firmata/ConfigurableFirmata/commit/f1b40e92f86785958f4295036c75bde62354888e

However if the disconnection state is not know immediately (or within a few ms) then this may still overrun the buffer.

jnsbyr, 2016-01-17:
The problem with the delayed detection of the disconnect cannot be helped when using TCP. My typical approach is using a periodic telegram to make disconnects obvious to both sides. This may be a possible future enhancement for Firmata but a detection within milliseconds cannot be guaranteed this way. 

What happens in Firmata when the serial read buffer is overrun? In my opinion it is not critical if "only" data is lost as long as no dynamic memory allocation is used and no memory corruption occurs. In this situation I would prefer to look into the TCP transmit queue size. But that means looking into the LAN chip driver source as far as I can tell.

Just ordered another Teensy to make testing easier and I have a spare I2C device to make my new test setup almost identical to my current setup. My idea is to link serial port 2 with port 3. Then I can create a steady data stream and receive it back.

soundanalogous, 2016-01-17:
> What happens in Firmata when the serial read buffer is overrun?

Currently nothing. There are no checks in Firmata. There may be some handling at the transport layer level, but nothing has ever been added in Firmata to address this... hence the outstanding comment about avoiding buffer overrun.

soundanalogous, 2016-01-17:
Another thing I could try as is if the Serial read buffer is full I could either discard the entire buffer or I could try to flush it (which would attempt to transmit it to the host). This would be performed each iteration of the main Arduino loop.

jnsbyr, 2016-01-17:
In normal operation it is preferrable to flush the serial buffer to the
host. But if the uplink send buffer is also full, discarding the buffer is
the rigth solution. Again one needs to know about the state of the uplink
buffer to make the right decision.

PaulStoffregen, 2016-01-17:
Recently I've been working on a redesign of the Arduino Ethernet library.  Much of the work has been keeping info on the Arduino/Teensy side about the status of the each socket in the Wiznet chip.  The main motivation is to remove the tremendous amount of unnecessary SPI communication that re-reads that state for every function call.  But it could also be used to allow programs to know how much data can be written to the buffer.  Serial now has availableForWrite().  Maybe it's time to do this on Ethernet too?

Whether Arduino will ever use my improvements is a good question.  They do add more code, and more RAM usage.  On larger ARM and larger AVR chips, that's a non-issue, but it could be an obstacle for memory limited boards like Arduino Uno.

soundanalogous, 2016-01-17:
`availableForWrite()` definitely sounds useful for Ethernet (and WiFi). Paul, do you know if repeatedly attempting to send data via an Ethernet client connection (sending one or more bytes per iteration of the main Arduino loop) would cause the program running on the Arduino to crash if the Ethernet client TX buffer is full due to the TCP connection being lost and the disconnect state not yet known by the sketch?

PaulStoffregen, 2016-01-17:
Ethernet.write() will wait.  So depending on how you define "crash", yes.

jnsbyr, 2016-01-18:
If Ethernet.write() blocks when the send buffer is full, it explains a part of the behaviour I have observed. 

But what I was missing was that normal operation was always resumed after the disconnect happened (netstat shows TIME WAIT or nothing for the session in question). This was not the case in some of my tests. Once or twice I waited an hour or more but to no avail. With the W5100 there seem to be to possible outcomes:

1) An explicit  client session close on the TCP server side during normal operation is successfully recovered by the client reconnecting.

2) Firmata seems to hang after several hours of normal operation (no session close this time but maybe a singular few seconds delay in reading the receive buffer on the host side). Firmata does not send anything to the host any more but is still pingable. 

The initial reason for case 2 this is still a mystery to me. I will try to setup a tcpdump with logging to get more data.

PaulStoffregen, 2016-01-18:
Here's the actual code deep within the Ethernet library.

https://github.com/arduino/Arduino/blob/master/libraries/Ethernet/src/utility/socket.cpp#L129

As you can see, that loop (starting at line 129) waits for either the W5100 chip to have enough room in its buffer, or for the connection to be in any state other than ESTABLISHED or CLOSE_WAIT.

Whether your application is really getting stuck forever in that loop is a good question.  If you want to track this down, perhaps edit that loop and add some code to toggle a pin or do something else you can observe.

jnsbyr, 2016-01-18:
Thanks for this insight. Your suggestion to toggle a pin in the loop is a good idea. In fact I will use at least two pins, one for the main Firmata loop and one for the loop in the socket send function. I will also add a serial write to the Firmata loop so that the serial stream is not interrupted by a TCP disconnect.

On the other hand the code in socket could be part of the possible solution I was looking for in the first place. By checking ```W5100.getTXFreeSize(s)``` on the Firmata layer it would be possible to check the buffer before calling ```send```. If there is not enough space available the serial data could be buffered at the Firmata layer or simply discarded. 

The problem here is as expected: that kind of solution is not LAN chip driver independent. A better solution would be to make Ethernet support a non-blocking version of ```send```. This could return the number of bytes that could be sent and leave the rest to the application the same way a regular socket does.

PaulStoffregen, 2016-01-18:
Non-blocking has been discussed over and over again.  The official decision from the Arduino devs, after an incredible amount of debate, was to keep write() blocking, but add availableForWrite() to the official API.  Well, at least for serial.  Eventually it'll probably come to Ethernet, especially since the socket layer has a function for it.

Just to be perfectly clear, the idea to make Arduino's write() function non-blocking has been discussed over and over again by many people.  The official decision was made.  Bringing it all up yet again isn't going to do anyone any good.  But there should be support for adding availableForWrite to the non-Serial libs.  If you lobby Arduino for changes, I highly recommend requesting availableForWrite() like Serial, rather than trying to convince them to change write().

jnsbyr, 2016-01-19:
No good beating a dead horse - that wasn't my intention, I just didn't know the background. 

Firmata uses ```EthernetClientStream``` to read and write on ```EthernetClient```. ```EthernetClient::write()``` uses ```socketSend()``` and always tries to send all bytes of the buffer or fails - there is no in-between. So one would have add ```availableForWrite()``` to ```EthernetClientStream```, ```EthernetClient``` and to ```socket``` to make it available in the application?

But first things first. I hope to get the first results with my new test setup by the weekend.

soundanalogous, 2016-01-19:
EthernetClientStream is a subclass of Stream that wraps EthernetClient. If `availableForWrite() is added to EthernetClient then it will be available to EthernetClientStream (and thus to Firmata). Ideally `availableForWrite()` would also be added to the Stream class as a virtual method.

jnsbyr, 2016-01-19:
Yes, but I it depends where and how you want to use ```availableForWrite()```. If you use it in ```Firmata.write()``` you get a generic solution that works for all transmits the same way but cannot tell the caller that the byte was not transmitted because the method is void. If you want to do something specific when not enough space is available (e.g. check in ```SerialFirmata``` to discard the serial input) you need to make ```availableForWrite()``` available through ```Firmata```.

PaulStoffregen, 2016-01-19:
> Ideally availableForWrite() would also be added to the Stream class as a virtual method.

Yes, that would be ideal.

After it's in Ethernet and a few more of the major libraries, my hope is the Arduino devs will consider that change.  Historically, they've been very conservative, to the point of getting almost any change nearly impossible.  Often it's taken many months, sometimes even years, to get them to warm up to an idea.  Just as things seemed to start improving, this war between Arduino.cc and Smart Projects / Arduino.org broke out, and things are even harder than before.  :(

jnsbyr, 2016-01-21:
Today communication was lost again between Firmata and host after about 4 days of continuous operation with one serial port enabled on the Teensy LC but no actual serial IO (the serial device was not connected). netstat showed that the connection was still active but the transmit buffer on the host was significantly filled.

It seems to me that serial IO only speeds up the point of instability but it is not the primary cause when using a Teensy LC with a W5100 LAN module. A memory leak connected with the amount of data transferred could be a possible reason. I have switched back to an ENC28J60 LAN module and keep the serial device disconnected to make a comparison.

soundanalogous, 2016-01-21:
Were you able to log the available memory value I exposed in the issue-37 branch? The available memory on the heap should be reported to your host application every 10 minutes via the SERIAL_DATA message ((the reporting interval is also configurable in the issue37.ino sketch).

soundanalogous, 2016-01-21:
STRING_DATA I mean... not SERIAL_DATA

jnsbyr, 2016-01-21:
Sorry, but I wasn't able to change the firmware of my setup last weekend. With the 2nd Teensy that arrived yesterday I will be able to do tests with your debug version on my desk and with my desktop computer. Even debugging via serial host interface will be possible this way. I hope that the problem can still be reproduced when using serial loopback from one serial port to another instead of an external serial device.

PaulStoffregen, 2016-01-21:
Did you get the pin toggling into the code?  Any confirmation of which loop was running when it got stuck?

jnsbyr, 2016-01-31:
I got the gut feeling that a part of the problems I have experienced so far is not directly related to Firmata or serial IO and has more to do with the hardware I used (Teensy LC + no-name [W5100](http://www.ebay.com/itm/Mini-W5100-LAN-Ethernet-Shield-Network-Module-board-for-Arduino-high-quality-/141353577912) or [ENC28J60](http://www.ebay.com/itm/ENC28J60-Ethernet-LAN-Network-Module-For-Arduino-SPI-AVR-PIC-LPC-STM32-DIY-/171351222608) modules). 

Before I go into debugging as you have suggested, I wanted to rebuild my setup on a breadboard to make debugging easier. I tried:
* Teensy LC 3.3 V + W5100 LAN module + Adafruit TS2561 I2C device 
  * failed to get a stable LAN communication when Teensy and W5100 are powered with 5.1 V, data is mostly garbled from the beginning
* Sparkfun Arduino Pro Mini 8 MHz 3.3 V + W5100 LAN module + Adafruit TS2561 I2C device 
  * failed to get a stable LAN communication when Arduino and W5100 are powered with 5.1 V, data is mostly garbled from the beginning
* no-name Arduino Pro Mini 16 MHz 5 V + W5100 LAN module + Adafruit TS2561 I2C device 
  * failed to get a stable LAN communication when Arduino RAW and W5100 are powered with 5.1 V, data is mostly garbled from the beginning
  * stable if Arduino VCC and W5100 are connected to 4.75 V (5.1 V + Schottky diode)
  * stable if Arduino RAW is connected to 5.1 V and W5100 is connected to Arduino VCC (app. 4.3 V)

These test results show that the power supply voltage for CPU and LAN are more critical to stability than can be assumed when reading the specifications. I was not able to make my original setup work on the breadboard. I cannot rule out that my LAN module is slightly degraded in respect to pin voltages but it is not very probable because according to specifications the W5100 chip is 5 V tolerant and there is an on-board 3.3 V regulator on my module. If you have any suggestions how to properly combine my W5100 LAN module with the Teensy LC or the Sparkfun Pro Mini 3.3 V please let me know.

Currently I use the Teensy LC without Firmata to generate continuous serial output for the Arduino with Firmata and SoftwareSerial (because the Pro Mini has only one hardware serial port exclusive for host communication). This setup has the advantage of a visual control: when the Teensy transmit I make its LED blink and as long as the Arduino works properly, the Arduino LED dimms, while the Arduino forwards the serial data to the host. This pattern changes after a while if the LAN cable is disconnected.

I had to modify the default StandardFirmataEthernetPlus example to enable serial support (the Arduino Firmata example should be updated to make it easier to use):
```C
65c65
< #if defined(ARDUINO_ARCH_AVR) || (ARDUINO >= 100 && ARDUINO < 10500)
---
> #if defined(ARDUINO_ARCH_AVR) || (ARDUINO >= 100 && ARDUINO < 10700)
519a520,522
>     case PIN_MODE_SERIAL:
>       // nothing to do
>       break;
807a811,814
>         }
>         if (IS_PIN_SERIAL(pin)) {
>           Firmata.write(PIN_MODE_SERIAL);
>           Firmata.write(getSerialPinType(pin));
```
This setup was stable in the last 24 hours. Reconnecting works, even if the software serial receive buffer gets filled by unplugging the LAN cable from the LAN-module for a few minutes. I will keep it running to check the long term stability.

soundanalogous, 2016-01-31:
I assume this change:
```
#if defined(ARDUINO_ARCH_AVR) || (ARDUINO >= 100 && ARDUINO < 10700)
```
Was only necessary to get SoftwareSerial to work with the Teensy LC? The Pro Mini should work fine without that change. However there is an issue in that no SAM or SAMD boards will compile if you make the above change since they do not include SoftwareSerial in their respective packages. I need to include the Teensy arch rather than include Arduino 1.6, or exclude SAM and SAMD. The other 2 issues were an oversight and were fixed a while back when I extracted the Serial feature into a [separate class](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.cpp).

jnsbyr, 2016-01-31:
Thank for the feedback. I was not sure about the ``` && ARDUINO < 10700``` when I switched from hardware serial to software serial and changed it just to be sure it's included. That was before I found out that the other 2 changes where the show stoppers. I took the code from ConfigurableFirmata-SerialFirmata.cpp so that should complete the backtrace :-).

As you said, SoftwareSerial depends mostly on the architecture so ```ARDUINO_ARCH_AVR``` should be sufficient for the Pro Mini. I don't want to interrupt my stability test, but I will confirm this as soon as possible. 

Sadly I cannot test hardware serial with the Pro Mini. But if someone can give me a hint how to setup a Teensy LC, my W5100 LAN module and a 5 V power supply with long term stability I could repeat the test with the Arduino as serial stream generator.

soundanalogous, 2016-01-31:
I think this change will include all of the Teensy boards but I'm not sure it's worth making since all Teensy boards have at least 1 UART available (Serial1) besides the USB connection.
```
#if defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_ARC32) || (ARDUINO >= 100 && ARDUINO < 10500) || defined(CORE_TEENSY)
#include <SoftwareSerial.h>
#endif
```

SoftwareSerial can cause all sorts of unexpected behaviors (such as if digital input is on the same port as the pins used for software serial). I debated whether or not I should even include it when I was working on including the Serial feature for Firmata but figured it was useful for AVR boards at least since the Uno is so prevalent. However users may still encounter various issues due to SoftwareSerial's various limitations.

jnsbyr, 2016-01-31:
Probably my description was not clear enough at this point. IMHO there should be no need for SoftwareSerial when using a Teensy (LC) with 4 hardware serial ports - and it is always preferrable to use hardware serial instead of software serial.

For the Pro Mini SoftwareSerial is the only way. When using 'only' one software serial port at 9600 baud there seem to be no critical limitations in regard to receiving data - but I havn't tested sending jet.

PaulStoffregen, 2016-02-01:
@soundanalogous - Email me directly (paul at pjrc dot com) and I'll send you hardware for testing.

jnsbyr, 2016-02-21:
A comparison of the behaviour between the Arduino Pro Mini ATmega 328 16MHz 5V and 8MHz 3.3V in combination with the popular Funduino W5100 Ethernet module had the following results:
* significant crosstalk occurs on the SPI bus when using a 50 mm ribbon cable between the Arduino and the 10 pin header of the Ethernet module
* signal distortion is critical for the 3.3V Arduino version, especially on MISO (e.g. extra spikes around 1V), causing data corruption and/or hang
* continuous serial data receive just increases the probability of a failure
* slight improvements are possible when using a 1k series end termination on MISO but this does not result in reliability
* the behaviour of the Teensy LC is similar to the Arduino Pro Mini ATmega 328 8MHz 3.3V
* the crosstalk level does not change significantly when using the 5V Arduino version, but the L/H decision level on the Arduino side is almost 1V higher - no instability was observed in more than 2 weeks of continuous operation in two different setups, but I will not claim that Arduinos at 5V are immune
* an application like Firmata on a 5V Arduino version can still be made to hang, e.g. by stopping the host application or disconnecting the Ethernet

Recommendations:
* do not use the Funduino W5100 Ethernet with a 3.3V Arduino/Teensy - the same probably applies to other boards with ribbon cable pin headers for the SPI bus that do not have GND on every 2nd pin
* always enable the WDT (Arduino) or COP (Teensy), regardless of the operating voltage - e.g. watchdog will trigger if Ethernet is disconnected, interrupting the transmit loop

Extras:
* the hardware serial port 0 (pins 0 and 1) of the Arduino Pro Mini ATmega 328 is available for Firmata when using Ethernet - instead of always excluding this port the port actually used by Firmata for host communication should be excluded
* there is no noticeable difference between hardware serial and software serial when receiving data on the Arduino Pro Mini ATmega 328 at 9600 baud
* timed store&forward with serial receive results in message transfers instead of single byte transfers significantly offloading the receiver on the host
* watchdog functionality can be further improved using a receive timeout check based on ```Firmata.available()```
* it is possible to connect to a serial Firmata device through an Ethernet Firmata device

I can provide patches for serial port 0, serial store&forward and WDT usage with StandardFirmataEthernetPlus and ConfigurableFirmata. 

From my point of view the instability issue observed is not Firmata specific. There are several reason why an Arduino setup with an Ethernet module can become unstable, possible causes being incompatible hardware, EMI/wiring and software error recovery (e.g. bus errors on SPI with W5100/ENC28J60, Ethernet disconnects). Enabling the watchdog can be a solution.

soundanalogous, 2016-02-21:
Regarding:

> the hardware serial port 0 (pins 0 and 1) of the Arduino Pro Mini ATmega 328 is available for Firmata when using Ethernet - instead of always excluding this port the port actually used by Firmata for host communication should be excluded

Are you talking about this specifically? https://github.com/firmata/arduino/issues/193

jnsbyr, 2016-02-21:
Yes, exactly - haven't check the issue list, just wanted to report that it is working.

soundanalogous, 2016-02-21:
Can you open a separate issue regarding the WDT usage? Do you have a sense
for whether or not that will work for all Arduino board architectures?

Regarding serial store & forward. I'm curious to see your solution here.
Either open a separate issue or a pull request.

thanks!

jnsbyr, 2016-02-28:
I committed the changes for serial store & forward to a new branch named "serial" based on current "master" but I do not have sufficient permissions to push this new branch to create a pull request.

soundanalogous, 2016-02-28:
You need to create a fork then create a branch off your fork and then create a pull request from that branch.