firmata/ConfigurableFirmata #44: Network configuration for Heroku

finfort, 2016-02-08:
I want to control my Ethernet shield from heroku hosting. That work great at local network, but what will be remote_ip and remote_port for heroku?
I have this dns https://arduinorelay.herokuapp.com/ but ip adress is dynamical and what about ports on heroku?
``` c 
#define NETWORK_FIRMATA

#define remote_ip IPAddress(192, 168, 0, 157)
#define remote_port 3030
#define local_ip IPAddress(192, 168, 0, 150)
const byte mac[] = {0x90, 0xA2, 0xDA, 0x0D, 0x07, 0x02};
#endif
```

``` js
var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});

board.on("ready", function() {
  var led = new five.Led(8);
  led.blink(500);
});
```

soundanalogous, 2016-02-08:
I haven't tried this but theoretically it should work:

1. Use `remote_host` rather than `remote_ip` and define the char string: `#define remote_host https://ardunorelay.herokuapp.com`

I'm not sure if https will be an issue or not.

Regarding the port, this is something you would define in your app running on heroku. If you are using node.js on the server for example, define a port to listen on then make sure this port is defined as the `remote_port` in ConfigurableFirmata.ino

soundanalogous, 2016-02-09:
It doesn't look like https will work: http://arduino.stackexchange.com/questions/4/how-to-get-https-on-arduino with the Ethernet library.

finfort, 2016-02-09:
If https doesn't work, how to control Arduino from hosting trough Ethernet?
Got this from heroku

``` code
2016-02-09T09:35:40.840122+00:00 heroku[web.1]: Starting process with command `node ./app/server/app.js`
2016-02-09T09:35:42.673876+00:00 app[web.1]: Express server listening on port 58376 in production mode
2016-02-09T09:35:42.734976+00:00 app[web.1]: 1455010542734 Connected Listening on port: 3030  
2016-02-09T09:35:42.732774+00:00 app[web.1]: 1455010542732 SerialPort Listening on port: 3030  
2016-02-09T09:35:42.735837+00:00 app[web.1]: Waiting for connection
2016-02-09T09:35:43.095342+00:00 heroku[web.1]: State changed from starting to up
2016-02-09T09:35:39.519638+00:00 heroku[web.1]: State changed from crashed to starting
2016-02-09T09:35:52.740334+00:00 app[web.1]: 1455010552740 Device or Firmware Error A timeout occurred while connecting to the Board. 
2016-02-09T09:35:52.740338+00:00 app[web.1]: 2016-02-09T09:35:52.740338+00:00 app[web.1]: 
2016-02-09T09:35:52.740339+00:00 app[web.1]: Please check that you've properly flashed the board with the correct firmware.
2016-02-09T09:35:52.740340+00:00 app[web.1]: See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
2016-02-09T09:35:52.740341+00:00 app[web.1]: 2016-02-09T09:35:52.740341+00:00 app[web.1]: 
2016-02-09T09:35:52.740342+00:00 app[web.1]: If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
2016-02-09T09:35:52.742661+00:00 app[web.1]: events.js:159
2016-02-09T09:35:52.742662+00:00 app[web.1]:       throw err;
2016-02-09T09:35:52.742663+00:00 app[web.1]:       ^
2016-02-09T09:35:52.742664+00:00 app[web.1]: Error: Uncaught, unspecified "error" event. ([object Object])
2016-02-09T09:35:52.742665+00:00 app[web.1]:     at Board.emit (events.js:157:17)
2016-02-09T09:35:52.742666+00:00 app[web.1]:     at Board.log (/app/node_modules/johnny-five/lib/board.js:636:8)
2016-02-09T09:35:52.742667+00:00 app[web.1]:     at Board.(anonymous function) [as error] (/app/node_modules/johnny-five/lib/board.js:647:14)
2016-02-09T09:35:52.742668+00:00 app[web.1]:     at Board.<anonymous> (/app/node_modules/johnny-five/lib/board.js:414:14)
2016-02-09T09:35:52.742669+00:00 app[web.1]:     at Timer.listOnTimeout (timers.js:92:15)
2016-02-09T09:35:53.389447+00:00 heroku[web.1]: State changed from up to crashed
2016-02-09T09:35:53.342174+00:00 heroku[web.1]: Process exited with status 1
```

Sketch using Nano v3 and  ENC28J60 
``` c
#include "ConfigurableFirmata.h"
#include <UIPEthernet.h>
#define remote_host  "https://arduinorelay.herokuapp.com"
#define remote_port 3030
#define local_ip IPAddress(192, 168, 0, 150)
const byte mac[] = {0x90, 0xA2, 0xDA, 0x0D, 0x07, 0x02};
```

finfort, 2016-02-09:
Is that connected somehow with port forwarding? https://youtu.be/bNENUbDxYZA?t=454

soundanalogous, 2016-02-09:
http should work without issue, Its https specifically that I'm not sure about. One way to find out is to run the [TcpClient](https://github.com/ntruchsess/arduino_uip/blob/master/examples/TcpClient/TcpClient.ino) example that is included with the UIPEthernet library. Uncomment line 50 and set an "http" address. You should get a response. Then try again with an "https" address.

Try to get the networking aspect working in this way before adding Firmata. It will be easier to debug. You are getting into uncharted territory with ConfigurableFirmata as most people only use it on a local network (including myself).

soundanalogous, 2016-02-09:
Regarding port forwarding, are you talking about creating a proxy to your web application so the user gets static files from port 80 or 8080 and the app is exposed via a proxy to port 3030. You can achieve this with an NGINX server that is a static content server and forwards to a node app running on the same server. Google "nginx node app proxy" and you'll find some tutorials on how to set this up.

finfort, 2016-02-10:
Reason that i use Firmata with internet because i want to give access to user to control my arduino remotely, not only in my network, i thought that in future there will be nodejs with authentication on hosting, with johnny five, mobile, web, etc. Maybe there should be local server and remote server on hosting to work properly, but that not really what i want. 
Tcp client that you propose works well with server.local, http and https as well
client.connect("http://arduinorelay.herokuapp.com/",5000)
client.connect("https://arduinorelay.herokuapp.com/",5000)
I always got client connected from serial monitor, so configurable firmata also should work? Thinking now how to debug it...
And are there any possibilities to get data from TcpClient with johnny five to ensure that it works with heroku

soundanalogous, 2016-02-11:
If you're getting a response packet when calling client.connect("https://...") then the connection should be working.

Try increasing [reportVersionTimeout](https://github.com/jgautier/firmata/blob/master/lib/firmata.js#L412) to a much higher value.

I'll try to find some time this weekend to set up a test case on a remote server.

soundanalogous, 2016-02-11:
I tested on a remote server and was able to get a simple example working without issue. I did discover a couple of things however:

- Don't include "http://" or "https:" in the `remote_host` definition, so define as:
```c++
#define remote_host  "arduinorelay.herokuapp.com"
```
- Make sure you define a local IP and it should be on your local subnet. It did not work with DHCP.

Here's my test code (running on a remote server):
```javascript
var Firmata = require("firmata").Board;
var EtherPort = require("etherport");
var board = new Firmata(new EtherPort(3030));

board.on("ready", function() {
  console.log("READY!");
  console.log(
    board.firmware.name + "-" +
    board.firmware.version.major + "." +
    board.firmware.version.minor
  );

  var state = 1;

  this.pinMode(8, this.MODES.OUTPUT);

  setInterval(function() {
    this.digitalWrite(8, (state ^= 1));
  }.bind(this), 500);

  this.analogRead(0, function(value) {
    console.log(value);
  });
});
```

For the server, I'm simply running Ubuntu 14.04 in on a DigitalOcean droplet (a VM).

On the Arduino, I'm running the following sketch on an Arduino Uno with an ENC28J60 shield attached. The led blinks on my board and the analog input is output to the console on the remote server.

```c++
/*
 * EthernetHost.ino generated by FirmataBuilder
 * Thu Feb 11 2016 01:05:05 GMT-0500 (EST)
 */

#include <ConfigurableFirmata.h>

#include <UIPEthernet.h>
#include <EthernetClientStream.h>

EthernetClient client;

const byte mac[] = {0xDE, 0xAA, 0xBB, 0xCC, 0xDD, 0x01};
int remotePort = 3030;
IPAddress localIp(10, 0, 0, 50);
#define REMOTE_HOST "myremotehost.com"
EthernetClientStream stream(client, localIp, IPAddress(0, 0, 0, 0), REMOTE_HOST, remotePort);

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;

#include <FirmataExt.h>
FirmataExt firmataExt;

#include <AnalogWrite.h>

#include <FirmataReporting.h>
FirmataReporting reporting;

void systemResetCallback()
{
  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_ANALOG(i)) {
      Firmata.setPinMode(i, ANALOG);
    } else if (IS_PIN_DIGITAL(i)) {
      Firmata.setPinMode(i, OUTPUT);
    }
  }
  firmataExt.reset();
}

void setup()
{
  Ethernet.begin((uint8_t *)mac);
  Firmata.setFirmwareVersion(FIRMWARE_MAJOR_VERSION, FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);

  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_SPI(i)
        || 4 == i
        || 10 == i
#if defined(__AVR_ATmega32U4__)
        || 24 == i
        || 28 == i
#endif
      ) {
      Firmata.setPinMode(i, PIN_MODE_IGNORE);
    }
  }

  pinMode(PIN_TO_DIGITAL(4), OUTPUT);
  digitalWrite(PIN_TO_DIGITAL(4), HIGH);

#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
  pinMode(PIN_TO_DIGITAL(53), OUTPUT);
#endif

  Firmata.begin(stream);

  systemResetCallback();
}

void loop()
{
  digitalInput.report();

  while(Firmata.available()) {
    Firmata.processInput();
  }

  if (reporting.elapsed()) {
    analogInput.report();
  }
}
```

soundanalogous, 2016-02-11:
Also make sure you run `npm install firmata` and `npm install etherport` on your server. Simply copying over the prebuilt modules may not work.

finfort, 2016-02-15:
I can't create DigitalOcean account to test it, debit card issues, but great that it works though,  i open'd a Ticket #333306 on heroku and wait for their response,  now got this in logs..
```
2016-02-11T08:48:06.446244+00:00 heroku[web.1]: State changed from up to starting
2016-02-11T08:48:09.867523+00:00 app[web.1]: Express server listening on port 16574 in production mode
2016-02-11T08:48:10.299717+00:00 heroku[web.1]: State changed from starting to up
2016-02-11T08:48:12.358667+00:00 heroku[web.1]: Stopping all processes with SIGTERM
2016-02-11T08:48:15.168043+00:00 heroku[web.1]: Process exited with status 143
2016-02-11T09:23:09.266036+00:00 heroku[web.1]: Idling
2016-02-11T09:23:14.263389+00:00 heroku[web.1]: Process exited with status 143
```

finfort, 2016-02-15:
So i am testing your firmware, i think i already get rid from 
``` txt
Stopping all processes with SIGTERM
``` 
heroku warning and now i have this log 
``` txt
2016-02-15T08:03:16.131301+00:00 heroku[web.1]: Starting process with command `node ./app/server/app.js`
2016-02-15T08:03:19.409921+00:00 app[web.1]: Express server listening on port http://localhost:36833 in production mode
2016-02-15T08:03:19.420619+00:00 app[web.1]: 1455523399420 Connected Listening on port: 3030  
2016-02-15T08:03:19.417246+00:00 app[web.1]: 1455523399416 SerialPort Listening on port: 3030  
2016-02-15T08:03:19.422469+00:00 app[web.1]: Waiting for connection
2016-02-15T08:03:19.900960+00:00 heroku[web.1]: State changed from starting to up
2016-02-15T08:03:29.426896+00:00 app[web.1]: 1455523409426 Device or Firmware Error A timeout occurred while connecting to the Board. 
2016-02-15T08:03:29.426899+00:00 app[web.1]: 2016-02-15T08:03:29.426899+00:00 app[web.1]: 
2016-02-15T08:03:29.426901+00:00 app[web.1]: Please check that you've properly flashed the board with the correct firmware.
2016-02-15T08:03:29.426902+00:00 app[web.1]: See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
2016-02-15T08:03:29.426902+00:00 app[web.1]: 2016-02-15T08:03:29.426902+00:00 app[web.1]: 
2016-02-15T08:03:29.426903+00:00 app[web.1]: If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
2016-02-15T08:03:29.453119+00:00 app[web.1]: Board Error:
2016-02-15T08:03:29.453123+00:00 app[web.1]:   { type: 'error',
2016-02-15T08:03:30.407153+00:00 heroku[web.1]: State changed from up to crashed
2016-02-15T08:03:30.387163+00:00 heroku[web.1]: Process exited with status 1
```
With that firmata
``` c
#include <ConfigurableFirmata.h>

#include <UIPEthernet.h>
#include <EthernetClientStream.h>

EthernetClient client;

const byte mac[] = {0xDE, 0xAA, 0xBB, 0xCC, 0xDD, 0x01};
int remotePort = 3030;
IPAddress localIp(192, 168, 0, 150);
#define REMOTE_HOST "arduinorelay.herokuapp.com"
EthernetClientStream stream(client, localIp, IPAddress(0, 0, 0, 0), REMOTE_HOST, remotePort);

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;

#include <FirmataExt.h>
FirmataExt firmataExt;

#include <AnalogWrite.h>

#include <FirmataReporting.h>
FirmataReporting reporting;

void systemResetCallback()
{
  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_ANALOG(i)) {
      Firmata.setPinMode(i, ANALOG);
    } else if (IS_PIN_DIGITAL(i)) {
      Firmata.setPinMode(i, OUTPUT);
    }
  }
  firmataExt.reset();
}

void setup()
{
  Ethernet.begin((uint8_t *)mac, localIp);
  Firmata.setFirmwareVersion(FIRMWARE_MAJOR_VERSION, FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);

  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_SPI(i)
        || 4 == i
        || 10 == i
#if defined(__AVR_ATmega32U4__)
        || 24 == i
        || 28 == i
#endif
      ) {
      Firmata.setPinMode(i, PIN_MODE_IGNORE);
    }
  }

  pinMode(PIN_TO_DIGITAL(4), OUTPUT);
  digitalWrite(PIN_TO_DIGITAL(4), HIGH);

#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
  pinMode(PIN_TO_DIGITAL(53), OUTPUT);
#endif

  Firmata.begin(stream);

  systemResetCallback();
}

void loop()
{
  digitalInput.report();

  while(Firmata.available()) {
    Firmata.processInput();
  }

  if (reporting.elapsed()) {
    analogInput.report();
  }
}
```
Code works on localhost...

finfort, 2016-02-15:
``` c
Ethernet.begin((uint8_t *)mac, localIp);
``` 
How you do this without localIp?

soundanalogous, 2016-02-15:
I see your trying to run Express as well. Try running only the following on heroku:

```javascript
var Firmata = require("firmata").Board;
var EtherPort = require("etherport");
var board = new Firmata(new EtherPort(3030));

board.on("ready", function() {
  console.log("READY!");
  console.log(
    board.firmware.name + "-" +
    board.firmware.version.major + "." +
    board.firmware.version.minor
  );

  var state = 1;

  this.pinMode(8, this.MODES.OUTPUT);

  setInterval(function() {
    this.digitalWrite(8, (state ^= 1));
  }.bind(this), 500);

  this.analogRead(0, function(value) {
    console.log(value);
  });
});
```

finfort, 2016-02-16:
You are right..I tried tried without Express and get this results(comments in scektch)
so still in confussion, and i can't try digital ocean...
``` c
#define remote_ip IPAddress(192, 168, 0, 157) // works (run node from local machine)
//#define remote_host  "arduinorelay.herokuapp.com" // didn't work (heroku node)
//#define remote_host  "server.local" // didn't work (run node from local machine)
```
```c
#include "ConfigurableFirmata.h"
#include <UIPEthernet.h>

#if defined ethernet_h || defined UIPETHERNET_H || defined _YUN_CLIENT_H_
#define NETWORK_FIRMATA

#define remote_ip IPAddress(192, 168, 0, 157) // works
//#define remote_host  "arduinorelay.herokuapp.com" // didn't work
//#define remote_host  "server.local" // didn't work
#define remote_port 3030
#define local_ip IPAddress(192, 168, 0, 150)
const byte mac[] = {0x90, 0xA2, 0xDA, 0x0D, 0x07, 0x05};
#endif

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;

//#include <Servo.h>
//#include <ServoFirmata.h>
//ServoFirmata servo;
// ServoFirmata depends on AnalogOutputFirmata
#if defined ServoFirmata_h && ! defined AnalogOutputFirmata_h
#error AnalogOutputFirmata must be included to use ServoFirmata
#endif

//#include <Wire.h>
//#include <I2CFirmata.h>
//I2CFirmata i2c;
//
//#include <OneWireFirmata.h>
//OneWireFirmata oneWire;
//
//#include <StepperFirmata.h>
//StepperFirmata stepper;

#include <SerialFirmata.h>
SerialFirmata serial;

#include <FirmataExt.h>
FirmataExt firmataExt;

//#include <FirmataScheduler.h>
//FirmataScheduler scheduler;

// To add Encoder support you must first install the FirmataEncoder and Encoder libraries:
// https://github.com/firmata/FirmataEncoder
// https://www.pjrc.com/teensy/td_libs_Encoder.html
// #include <Encoder.h>
// #include <FirmataEncoder.h>
// FirmataEncoder encoder;

/*===================================================================================
 * END FEATURE CONFIGURATION - you should not need to change anything below this line
[... 233 lines cut ...]
```

``` js 

var EtherPort = require("etherport");

var Firmata = require("firmata").Board;

var board = new Firmata(new EtherPort(3030));

board.on("ready", function () {
    console.log("Arduino connected using Firmata.js!");
    var state = 1;
    this.pinMode(6, this.MODES.OUTPUT);
    setInterval(function () {
        this.digitalWrite(6, (state ^= 1));
    }.bind(this), 500);
});
board.on("error", function (msg) {
    console.log("Board Error:\n ", msg);
    process.exit(1);
});

console.log('Waiting for connection');
```

soundanalogous, 2016-02-17:
You don't need to use DigitalOcean specifically. Any service that can get you access to a VM should work (such as Amazon Web Services), but you will need to know server administration basics if you take that route.

I have a Heroku account that I messed around with years ago. However I can't bet back in because I can't find my recovery code. I'll see if I can create a new account and if setup doesn't take too much time.

soundanalogous, 2016-02-17:
I created a new Heroku account. The app appears to start up on the Heroku server, but I cannot establish a connection from the Arduino. Maybe Heroku is blocking the port or something. I don't understand enough about Heroku. Your best bet is to find someone who has experience with node.js and Heroku since the issue does not appear to be in Firmata.

finfort, 2016-02-17:
Oh yes will think now specifically about hosting

finfort, 2016-02-17:
I'll able to setup aws hosting, open ports, but still there no connection from VM, but it's there on localhost)
Was using this variables 
``` c
#define remote_ip IPAddress(52, 91, 215, 224)
#define remote_host  "ec2-52-91-215-224.compute-1.amazonaws.com"
 ```
[image]

soundanalogous, 2016-02-17:
I tried the same code I was running on my DigitalOcean Ubuntu instand on a new AWS instance and it works. I provisioned a new Ubuntu server (a raw Ubuntu server - don't use Bitnami). Then enabled TCP on port 3030 for 0.0.0.0/0 just as you did.

I installed node and npm like this:

```bash
curl -sL https://deb.nodesource.com/setup | sudo bash -
sudo apt-get install nodejs
sudo apt-get install build-essential
```

Then created a folder in my home directory and within that folder ran:

```bash
npm install firmata
npm install etherport
```

Then created a new file `app.js` and added the same code I used for the DigitalOcean example.

On the Arduino, I used the public IP address for my AWS server.

Run `node app.js` on the server. There were a couple of `ECONNRESET` errors the first couple of times I hit reset on the board, but after that it started working, printing the analog input values in the console and blinking the LED on the board.

soundanalogous, 2016-02-17:
Sometimes you have to hit reset on the board and wait several seconds before the connection is made.

closed 2016-12-24 by soundanalogous