firmata/ConfigurableFirmata #45: Can't get PIN_MODE_PULLUP to work with johnny-five

ernstnaezer, 2016-02-15:
Hi.

I'm looking for pointers to debug an issue with a button in PIN_MODE_PULLUP mode.

The following code works with the Standard Firmata sample from Arduino:

```javascript
var five = require("johnny-five");
five.Board().on("ready", function() {
  var button = new five.Button({pin: 2, isPullup: true});
  button.on("down", function(value) { console.log('down') });
  button.on("up", function() { console.log('up') });
});
```
With ConfigurableFirmata I get one **up** and then silence. 

I've traced though both Standard and Configurable Firmata, but I can't find a real difference other then structure. 

Any pointers on how to debug this would be much appreciated. 

Both with the latest versions (cf master & johnny-five is at version 0.9.19)

soundanalogous, 2016-02-15:
node-firmata, which johnny-five uses to communicate with an Arduino board does not yet implement `PIN_MODE_PULLUP`: https://github.com/jgautier/firmata/blob/master/lib/firmata.js#L424-L438. However I believe the old method of setting a digital input to `HIGH` should work and that is what `isPullup: true` does behind the scenes: https://github.com/rwaldron/johnny-five/blob/master/lib/button.js#L32-L34.

What board are you using? What you are trying won't work on an Arduino Zero, but should work on other boards.

ernstnaezer, 2016-02-16:
Ah, that explains why I couldn't find anything that referenced to `PIN_MODE_PULLUP` in Johnny-five. Interesting.

I'm using an Arduino UNO with a tilt sensor attached to the board right like so: http://johnny-five.io/examples/button-pullup/

soundanalogous, 2016-02-16:
That's correct. Try removing `Firmata.getPinState(pin) != 1` from the condition. I wonder if state is somehow set to 1 in johnny-five.

ernstnaezer, 2016-02-16:
I tried your suggestion but unfortunately it makes no difference. The button code still not behaves as expected.

soundanalogous, 2016-02-16:
As a control, add a physical pull-up resistor and see if that works with the same code.

Also try on a different pin. I have seen internal pull-up resistors burn out on pins in the past.

I will try running it tonight if I have some time.

soundanalogous, 2016-02-17:
This works in StandardFirmata and StandardFirmataPlus but does not work in ConfigurableFirmata
- Arduino Uno
- Arduino IDE version 1.6.7
- StandardFirmata (from Firmata v2.5.2)
- also test with StandardFirmataPlus with no issue
- johnny-five v0.9.21
- pasted the exact same code you referenced above
- Wire one end of button to ground and the other to pin D2 on the Uno

Looking into the issue in ConfigurableFirmata...

soundanalogous, 2016-02-17:
Fixed: https://github.com/firmata/ConfigurableFirmata/releases/tag/2.8.2.

closed 2016-02-17 by soundanalogous

ernstnaezer, 2016-02-17:
Fantastic! That works like a charm.

The fix has the signature of a race condition. Also, I am I correct in stating that this change:

```
- Firmata.setPinMode(pin, INPUT_PULLUP); 
+ pinMode(pin, INPUT_PULLUP); 
```

is to avoid this:

https://github.com/firmata/ConfigurableFirmata/blob/master/src/DigitalInputFirmata.cpp#L96-L97

soundanalogous, 2016-02-17:
Yes, it was a tricky solution I had to come up with in order to support both the old way of enabling pull-ups and the new way simultaneously.

ernstnaezer, 2016-02-17:
Thanks for your explanation and time to fix this. Much appreciated.