firmata/ConfigurableFirmata #53: Do not include <SoftwareSerial.h> unless the user wants it.

merose, 2016-07-06:
In SerialFirmata.h there is a #include of <SoftwareSerial.h>. This is problematic because the latter file defines an interrupt handler for the Port B interrupt (on Uno and Leonardo). This definition will conflict with an ISR defines for another library, such as the EnableInterrupt library.

Specifically, I'd like to use ConfigurableFirmata together with the EnableInterrupt Arduino library to define my own Firmata feature for wheel encoders, using the pin-change interrupts exposed by EnableInterrupt. The only way to do that right now is to manually edit SerialFirmata.h to avoid including SoftwareSerial.h.

To reproduce:
- In the Arduino IDE, install ConfigurableFirmata using Sketch > Include Library > Manage Libraries.
- Also install the EnableInterrupt library.
- Load the attached sketch SoftwareSerialProblem.ino into the Arduino IDE.
- Try to compile the sketch using Tools > Verify/Compile.

Expected result: Sketch compiles and links without problems.
Actual result: Error linking the sketch because of multiple definitions of "__vector_9". See error output below.
[SoftwareSerialProblem.zip](https://github.com/firmata/ConfigurableFirmata/files/350958/SoftwareSerialProblem.zip)

    Arduino: 1.6.9 (Mac OS X), Board: "Arduino Leonardo"
    
    In file included from /Users/merose/Documents/Arduino/SoftwareSerialProblem/SoftwareSerialProblem.ino:17:0:
    /Users/merose/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:22:121: note: #pragma message: NOTICE: *** EnableInterrupt library version 0.9.5. This is not a problem. Keep calm, and carry on. ***
     #pragma message("NOTICE: *** EnableInterrupt library version 0.9.5. This is not a problem. Keep calm, and carry on. ***")
                                                                                                                             ^
    libraries/SoftwareSerial/SoftwareSerial.cpp.o: In function `__vector_9':
    /Users/merose/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:229: multiple definition of `__vector_9'
    sketch/SoftwareSerialProblem.ino.cpp.o:/Users/merose/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1577: first defined here
    collect2: error: ld returned 1 exit status
    exit status 1
    Error compiling for board Arduino Leonardo.
    
    This report would have more information with
    "Show verbose output during compilation"
    option enabled in File -> Preferences.

soundanalogous, 2016-07-06:
Do you have any ideas for a way around including SoftwareSerial.h? I'd probably have to split the serial feature into separate HW serial and SW serial versions.

soundanalogous, 2016-07-06:
Oh... actually I see what the problem is. It can be fixed by moving the SerialFirmata implementation into the header file. I'm finding I have to do this for pretty much all classes to avoid conflicts. It's an unfortunate issue with the way Arduino links and builds libraries.

soundanalogous, 2016-07-06:
Also, have you tried using the Firmata Quadrature encoder feature? https://github.com/firmata/FirmataEncoder

merose, 2016-07-07:
Hi Jeff,
Thanks for looking at this so quickly. I didn't try moving the code into the .h file. Anxious to try that out whenever you like.
I like FirmataEncoder, but wish it weren't hard-coded to use the Encoder library. I'm using a Pololu A-Star Arduino board, which is Leonardo-compatible with a 32U4 chip. I'm using the UART (pins 0 and 1) to talk to a Raspberry Pi (ROS-based higher-level control for the robot), and I'd also like to keep pins 2 and 3 free for I2C. That only leaves pin 7 for hardware interrupts, hence I want to use the EnableInterrupt library to get the encoder interrupts myself. I guess I'd like to have some sort of base Encoder class like this:
class EncoderBase {public:    EncoderBase(int pinA, int pinB, int useBothInterrupts=true);    virtual int32_t read() = 0;    virtual void write(int32_t value) = 0;    virtual void update() = 0;};

Then derive from that to make, say MyEncoder. Then if FermataEncoder were a template class, you'd have:
#include <FermataEncoder.h>FermataEncoder<MyEncoder> encoder;
(And it could include an implementation for hardware interrupts using the Encoder library, too, but you'd be able to avoid it if you wanted to use pin-change interrupts, as I do.)
Right now I'm using the ideas in FermataEncoder to roll my own Firmata extension.
-----
Another thing I'd like is a base class for a SYSEX extension, perhaps also templatized with the subcommand, like this:
template<int subcommand> class SysexCommandBase : FirmataFeature {public:    ...    boolean handleSysex(byte command, byte argc, byte *argv); // Implementation provided, which checks the subcommand and then calls:    virtual boolean handleSubcommand(byte argc, byte *argv) = 0;
    ...};
If I get something I like I'll run it by you.
Mark
--
Mark Rose
markrose@acm.org
925.922.7263

soundanalogous, 2016-07-15:
New plan. SerialFirmata will be moved out of the src directory as explained [here](https://github.com/finson/FirmataWithDeviceFeature/tree/b2.10.0/features-optional). This will prevent the Arduino IDE from trying to compile any of the non-core features.