firmata/ConfigurableFirmata #54: Need to add license information

merose, 2016-07-08:
There is no licensing information in ConfigurableFirmata. In the USA, without explicit licensing all rights are reserved by the copyright holders (the authors), essentially making the library illegal to use or modify without getting explicit permission from the copyright holders. It would be more convenient if there were a license file included in the repository, preferably one that's A) open-source friendly, and B) friendly for commercial use. (A) is more important than (B), of course, though being commercial-friendly would help Firmata gain wider adoption, IMHO.

Suggestions:
- If you don't object to a commercial-friendly license use the BSD (new, 3-clause) or the Apache 2.0 license, or even Creative Commons Zero if you want to put it in the public domain.
- If you object to commercial use, use GPL (but please consider this carefully, as it will likely limit adoption of Firmata).
- Don't use the Eclipse license, as I've had negative reactions from lawyers.

This issue also applies to the original Arduino Firmata.

soundanalogous, 2016-07-08:
It should have the same GNU license as the Arduino Firmata implementation because code is derived from that project: https://github.com/firmata/arduino/blob/master/extras/LICENSE.txt.

soundanalogous, 2016-07-08:
Actually it does have a license: https://github.com/firmata/ConfigurableFirmata/tree/master/extras

closed 2016-07-08 by soundanalogous

merose, 2016-07-08:
My mistake, sorry, didn't see it.

soundanalogous, 2016-07-08:
I could move the License file to the root dir to make it more discoverable. At one point, Arduino recommended putting it in Extras when they first published the v1.5 library guidelines, but perhaps that has changed. It makes more sense for it to be in the root anyway IMO.

merose, 2016-07-08:
Another option would be to talk about the license in the README in the root. (I.e., just have a "License" heading and say "This software is licensed under the terms of the GNU Public License. See extras/LICENSE.txt for details.) I think either that or putting the license file in the root would have prevented my confusion. (I may be atypical, but I generally read the whole README for licensing info, and if not there, look for a LICENSE file.)