firmata/ConfigurableFirmata #59: Using stepper deceleration breaks callback functionality

troywweber7, 2016-11-09:
Setting `decel` parameter in `stepperStep()` of firmata.js to anything but `0` breaks the callback functionality. It completes the run completes on the arduino fine, but its as if the javascript side never gets the proper response for which it is waiting. I sense that there is an issue involving these two lines of code (https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/FirmataStepper.cpp#L178, https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/FirmataStepper.cpp#L327), but I am uncertain at this point.

Does anyone have a fix for this? Can whomever wrote "FiramataStepper.cpp" or contributed to `ConfigurableFirmata` confirm why using the `decel` parameter might cause it to never respond? You pretty much can't use deceleration at all if this isn't fixed.

Thanks for any help/direction

pmoliterni, 2016-11-30:
Hi,
I have the same problem and I think I've found a solution. This happens only when you pass an acceleration/decelleration value to the **stepperStep** function, so I went inside the code of the finite-state machine which regulate the movement. The problem is due to the conversion from the interupt-based algorithm (the original one from ATMEL) to the Arduino version which is polling-based. During the last step, the variable **newStepDelay** in the file **FirmataStepper.cpp** becomes negative but it is declared **unsigned long** so the line **newStepDelay = -newStepDelay** causes the next step is really big.

I think a good way to solve the problem (without modifying the original algorithm and considering that with the interrupt based version the last step is computed in the previous calculation) is to copy the lines of code of the **STOP** state directly in the **DECEL** state (if you want even in the **RUN** state). This solved the problem for me. 

Otherwise you may avoid to update the new step if you are in the STOP state:

```
if(this->run_state == FirmataStepper:STOP)
step_delay = newStepDelay
```

With the last solution I guess you will have one more step which may compromise your computed distance

I hope the administrator may answer to this question and maybe modifying these lines of code or finding a better way to solve it.

Best Regards,
Pasquale

soundanalogous, 2016-11-30:
We are swapping out my custom Firmata stepper code for AccelStepper. Work is in progress.

pmoliterni, 2016-12-01:
I've seen the **AdvancedFirmata** and the variable **newStepDelay** is declared **long** instead of **unsigned long** as for **ConfigurableFirmata**. This solve the problem but I think it will compute one step more compared with the original code from ATMEL. In the ATMEL code the variable **newStepDelay** is declared **unsigned long** and even if it become negative (or better a big positive number) and it is assigned to **stepDelay**, this last big positive value will not be used because the state machine will immediately enter in the STOP state. Besides my suggestions above, another solution might be the following

`
if((this->running == true && timeDiff >= this->step_delay) || FirmataStepper::STOP)
`

Best Regards,
Pasquale.

soundanalogous, 2016-12-01:
@pmoliterni if you've found a solution you're confident with, please feel free to submit a pull request. Otherwise I'll try to look into it this weekend.

pmoliterni, 2016-12-01:
@soundanalogous Ok I'll try but because you are the author of the code, I'm sure you'll find a better solution. I hope my suggestions will help you through it.

Best Regards,
Pasquale.

soundanalogous, 2016-12-04:
Fixed: https://github.com/firmata/ConfigurableFirmata/releases/tag/2.9.2

closed 2016-12-04 by soundanalogous

pmoliterni, 2016-12-04:
@soundanalogous Great job. Thank you.