firmata/ConfigurableFirmata #60: stepperConfig causes failure of communication with Arduino

pmoliterni, 2016-12-02:
I've noticed a strange behaviour using **ConfigurableFirmata** (as is, without the use of firmatabuilder) and **firmata.js** with **node.js** (this happens also with Snap4Arduino). When I use both the following lines of code the communication with Arduino fails:
```
var DRV_4WIRE = 0x04;
var numStepsForRev = 64;
board.stepperConfig(0, DRV_4WIRE, numStepsForRev, 3, 5, 4, 6);   
board.stepperConfig(1, DRV_4WIRE, numStepsForRev, 9, 11, 10, 12);
```
If I use **only one of the two lines** the board works perfeclty. This behaviuor disappears if I comment the lines in **ConfigurableFirmata.ino** #172-#173 
```
//#include <I2CFirmata.h>
//I2CFirmata i2c 
```
**or** the lines #175-#176 
```
//#include <OneWireFirmata.h>
//OneWireFirmata oneWire. 
```
Is there something wrong with I2c or OneWire which causes the communication to fail ?

soundanalogous, 2016-12-02:
Are you using I2C or OneWire in your application?

pmoliterni, 2016-12-02:
No, I'm not using I2C or OneWire devices in my application. I've developed a simple test application with **firmata.js** and **node.js** which contains only the two stepConfig. I thought it might be a bug or something related with PIN_MODES or interrupts.

soundanalogous, 2016-12-02:
I'll look into it over the weekend. Can you post your node.js application code (perhaps in a gist) and link to it here?

pmoliterni, 2016-12-02:
Yes, of course.

```
var Board = require("../");

Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }

  var board = new Board(port.comName);

  board.on("ready", function() {
    var pin = 13;
    var state = 1;

    board.pinMode(pin, board.MODES.OUTPUT);

    setInterval(function() {
      board.digitalWrite(pin, (state ^= 1));
    }, 500);
    
    var DRV_4WIRE = 0x04;
    var numStepsForRev = 64;

    console.log("CONFIG START");

    board.stepperConfig(0, DRV_4WIRE, numStepsForRev, 3, 5, 4, 6);   
    board.stepperConfig(1, DRV_4WIRE, numStepsForRev, 9, 11, 10, 12); 
    
    console.log("CONFIG STOP");
    
	var CCW = 0,
	CW = 1;

	var steps_rev = 4076/2;
	var wheel_dia = 64.5;
	
	var distance = 50;
     
	var steps = distance * steps_rev / (wheel_dia * 3.141592654); //24.61							 
    
	var doneCallback0 = function () {
		console.log("CALL 0");
		//alert("CALL 0");           
		//done.add(++index);
	}
	
	var firmwareDone = function () {
		//console.log("FIRMWARE DONE ");
		//console.log("FIRMWARE NAME " + board.firmware.name);
		//console.log("FIRMWARE VERSION MAJOR" + board.firmware.version.major);
		//console.log("FIRMWARE VERSION MINOR" + board.firmware.version.minor);

		//alert("CALL 0");           
		//done.add(++index);
	}

	var doneCallback1 = function () {
		console.log("CALL 1");
		//alert("CALL 0");  
		//done.add(++index);
	}
	
	var doneCallbackNoAcc0 = function () {
		console.log("NO ACC 0");
		//alert("CALL 0");  
		//done.add(++index);
	}
	
	var doneCallbackNoAcc1 = function () {
		console.log("NO ACC 1");
		//alert("CALL 0");  
		//done.add(++index);
	}

	//board.stepperStep(0, CW, steps, 4000, 4000, 8000, doneCallback0);					
        //board.stepperStep(1, CCW, steps, 4000, 4000, 8000, doneCallback1);
	
	//board.stepperStep(0, CW, steps, 4000, doneCallbackNoAcc0);
	//board.stepperStep(1, CW, steps, 4000, doneCallbackNoAcc1);

	console.log("END");
	//board.queryFirmware(firmwareDone);
    
  });
});
```

soundanalogous, 2016-12-03:
@pmoliterni which Arduino board are you using?

pmoliterni, 2016-12-03:
@soundanalogous I am using Arduino Nano which should be the same as Arduino Uno but smaller

soundanalogous, 2016-12-03:
Okay the issue is you are running out of memory. Uno, Nano, Leonardo, etc don't have enough memory to include all of the ConfigurableFirmata features. That's why it works if you exclude OneWire and I2C.

soundanalogous, 2016-12-03:
If you look at the console in the bottom of the Arduino IDE, you should see the following message in red text when compiling for the Nano, Uno, Leonardo, etc:

`Low memory available, stability problems may occur.`

closed 2016-12-04 by soundanalogous

pmoliterni, 2016-12-04:
@soundanalogous Great job. Thank you.