firmata/ConfigurableFirmata #62: Not working without Serial USB connection?

nova76, 2016-12-15:
Hi
I got an Ethernet W5100 Shield (http://www.aeroboshop.com/.../ethernet-w5100-shield-module-f.../)
I like to use this fimata, so i was downloaded and uploaded to my arduino uno board.
If  i trying to ping the device,  i can get the answer until the USB is plugged in with a data cable. If the USB cable is replaced by only an power cord, than the arduino  is not connected to the network (can not be pinged) So, is not working this stuff without serial USB connection?

soundanalogous, 2016-12-15:
There is no reason it should not work with USB disconnected as long as the board is connected to a power source. I'll look into this further tonight to see if I'm able to reproduce the issue.

PaulStoffregen, 2016-12-15:
Especially if these are cheap clones, (carefully) use a voltmeter to check places on the ethernet board for power.  Many newer ethernet boards depends on IOREF voltage.  Older ones had they own issues, like depending on the reset pulse.

soundanalogous, 2016-12-16:
@nova76, Using an older official Arduino Ethernet Shield and Arduino Uno with a 9V power supply connected to the on-board barrel jack, I'm unable to reproduce the issue. Firmata over Ethernet works as expected. It's most likely an issue with your ethernet shield as Paul points out, especially if you purchased a cheap clone that doesn't break out the right headers.

Closing as this is a hardware issue rather than a software issue.

closed 2016-12-16 by soundanalogous

soundanalogous, 2016-12-16:
I will however start a wiki page to describe which Ethernet shields are known to be compatible with Firmata (this is largely TBD). Please let me know which specific shield you have.

soundanalogous, 2016-12-16:
Here's the wiki page I mentioned in the previous comment: https://github.com/firmata/ConfigurableFirmata/wiki/Compatible-Ethernet-shields

I'm sure there are many more compatible boards, I'm just not aware of them yet.