firmata/ConfigurableFirmata #66: Feature Request: Firmware reports limitations

troywweber7, 2016-12-28:
This is a minor feature request, but as I have been dealing with the stepper side of things, it has come to my attention that perhaps it would be useful if the firmware were to report the maximum number of steppers allowed, similar to how it reports the number of pins on the board and the limitations of said pins. In Stepper 1.0 protocol the max is 6. In Stepper 2.0 protocol, the max is 10. If someone daring enough (and I just might know someone) decides to run more than 10 steppers on an Arduino by changing the constant in the C++ code, then this constant should probably be reported to the firmata.js side to be used in issuing warnings or in error handling.

I would mark this as a `feature request` or an `enhancement`, but I don't know how.

soundanalogous, 2016-12-28:
I think this should be handled by the Firmata client (firmata.js in your case) rather than the firmware. The client developer should be familiar with the protocol and defined limitations.

soundanalogous, 2016-12-28:
Another more ambitious way to handle this would be to have the max number of steppers determined by available ram on the target microcontroller, then it would make sense for the firmware to report this value back. I'm not even sure that 10 steppers is realistic for all boards. It likely isn't for an Uno or Leonardo (2k and 2.5k RAM respectively) especially if several other features are enabled.