firmata/ConfigurableFirmata #70: example of how to use ConfigurableFirmataWifi

troywweber7, 2017-05-03:
I've been scraping around trying to figure out how to connect to my arduino over wifi with firmata. I uploaded the ConfigurableFirmataWifi.ino in this project (added the ssid and password, removed unneeded features so it takes up only %50 of the space on the Arduino Uno). The problem I'm having is connecting with the board from JavaScript.

I found an npm project `tcp-serial` as well as `udp-serial` or `tcpsocket-serialport` and none of their examples worked for me. The tcp-serial example looks like it is going to work board the board never responds with ready state.

Am I missing something? How does one connect to the ConfigurableFirmata over Wifi on the client side?

soundanalogous, 2017-05-03:
You will find two examples here: https://github.com/soundanalogous/firmata-client-examples/tree/master/firmata-js/network

See blink-firmata-client.js and blink-firmata-server.js

troywweber7, 2017-05-04:
Okay, so it behaves the same way as when I use `udp-serial` or `tcp-serial`, although I will stick with `etherport-client` since it is recommended in the examples. Unfortunately, my `'ready'` function never runs. My script is as follows:

```javascript
#!/usr/bin/env node

const EtherPortClient = require('etherport-client').EtherPortClient,
	Board = require('firmata').Board,
	repl = require('repl');

// get etherport client

let etherport = global.etherport = new EtherPortClient({ host: '192.168.1.7', port: 3030 });
 
// create the tcp serialport and specify the host and port to connect to

let board = global.board = new Board(etherport);

board.once('ready', () =>
{
	console.log('connected to board over wifi!');
});

repl.start();
```

My script is not all that different from the example (other than using more recent node.js/js syntax), so I'm assuming my issue might be on the Arduino side. I'll have to look over the file I uploaded again to make sure I followed all the instructions carefully, but I should ask this question: in the ConfigurableFirmataWifi it says that you *cannot* use an Arduino Uno because the full .ino file uses nearly all the space on the board. But if you remove features you don't need (i.e. Servo, Wire, OneWire, Serial, Ext, and Scheduler) then it only uses 50% of the space on the Arduino Uno and that should be fine, right? I'm using the legacy wifi shield, in case that matters.

soundanalogous, 2017-05-04:
I'm not sure if the legacy WiFi shield works since I've never had access to one to test. I only have the newer WiFi Shild 101 and know that works for sure (as does the MKR1000).

Make sure you have ConfigurableFirmataWiFi configured as a TCP Server. Also enable Serial degugging in the sketch (and connect to a Serial terminal - to view the debug output).

soundanalogous, 2017-05-04:
Also if you need a basic version of Firmata with WiFi, use [StandardFirmataWiFi](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi). That should run fine on an Uno.

troywweber7, 2017-05-04:
Hmm. I have an IoT with JS demo/talk to give on Tuesday, so I'm in a crunch (that's my bad, though, for putting this off for so long).

I doubt I'll have time to order it, but where did you purchase the Wifi Shield 101? I've look and have not been able to find one for purchase.

I'm going to try StandardFirmataWifi and see if I can get that to work. I was hoping to move a stepper motor, but I can scale back.

troywweber7, 2017-05-04:
So, just to confirm, I have ConfigFirmataWifi configured as a TCP Server and I get the following debug in the console.

```
ConfigurableFirmataWiFi will attempt a WiFi connection using the legacy WiFi library.
Using static IP: 192.168.1.100
Attempting to connect to WPA SSID: GS7
WiFi setup done
SSID: GS7
IP Address: 192.168.1.7
signal strength (RSSI): -42 dBm
TCP connection disconnected
```

troywweber7, 2017-05-04:
StandardFirmataWifi is also not working for me. I get the following debug messages.

```
StandardFirmataWiFi will attempt a WiFi connection using the legacy WiFi library.
IP will be requested from DHCP ...
Attempting to connect to WPA SSID: GS7
WiFi setup done
SSID: GS7
IP Address: 192.168.1.7
signal strength (RSSI): -47 dBm
TCP connection disconnected
```

In both cases, I am able to `ping 192.168.1.7` and get responses from the IP address.

```
troyw@tww-ubuntu:~$ ping 192.168.1.7
PING 192.168.1.7 (192.168.1.7) 56(84) bytes of data.
64 bytes from 192.168.1.7: icmp_seq=1 ttl=255 time=4.74 ms
64 bytes from 192.168.1.7: icmp_seq=2 ttl=255 time=62.0 ms
64 bytes from 192.168.1.7: icmp_seq=3 ttl=255 time=16.2 ms
64 bytes from 192.168.1.7: icmp_seq=4 ttl=255 time=74.8 ms
64 bytes from 192.168.1.7: icmp_seq=5 ttl=255 time=35.1 ms
64 bytes from 192.168.1.7: icmp_seq=6 ttl=255 time=167 ms
64 bytes from 192.168.1.7: icmp_seq=7 ttl=255 time=9.62 ms
64 bytes from 192.168.1.7: icmp_seq=8 ttl=255 time=42.7 ms
^C
--- 192.168.1.7 ping statistics ---
8 packets transmitted, 8 received, 0% packet loss, time 7008ms
rtt min/avg/max/mdev = 4.744/51.653/167.745/49.616 ms
```

troywweber7, 2017-05-04:
Also, using the script above, I moved the `repl.start()` into the `'ready'` callback and decided to let it set there for a while. I did finally get something (though not the something I was hoping fore).

```bash
$ cat board.js 
#!/usr/bin/env node

const EtherPortClient = require('etherport-client').EtherPortClient,
	Board = require('firmata').Board,
	repl = require('repl');

// get etherport client

let etherport = global.etherport = new EtherPortClient({ host: '192.168.1.7', port: 3030 });
 
// create the tcp serialport and specify the host and port to connect to

let board = global.board = new Board(etherport);

board.once('ready', () =>
{
	console.log('connected to board over wifi!');
	repl.start();
});
$ ./board.js 
(node:10310) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 connect listeners added. Use emitter.setMaxListeners() to increase limit
^C
```

soundanalogous, 2017-05-04:
I need to see all of your code to understand where the issue may be. Please post your full Arduino code (minus your password of course) to a gist (or elsewhere) and provide a link.

soundanalogous, 2017-05-04:
Also try removing the repl entirely (repl: false) in the Board constructor options.

troywweber7, 2017-05-04:
Full Arduino code:
- [StandardFirmataWifi scenario](https://bitbucket.org/troywweber/reno.js-intro-to-iot-with-js/src/6ffe9f4a120839aa1f32db53ed1aa06f601a1838/StandardFirmataWiFi/?at=master)
- [ConfigurableFirmataWifi scenario](https://bitbucket.org/troywweber/reno.js-intro-to-iot-with-js/src/6ffe9f4a120839aa1f32db53ed1aa06f601a1838/ConfigurableFirmataWiFi/?at=master)
- [board.js](https://bitbucket.org/troywweber/reno.js-intro-to-iot-with-js/src/6ffe9f4a120839aa1f32db53ed1aa06f601a1838/board.js?at=master&fileviewer=file-view-default)

I removed REPL entirely and it still does not work, although, when you talk about removing REPL in the Board constructor options, I believe you are thinking of Johnny-Five's implementation of a board. Here we are just using straight firmata board.

Thanks for your help on this. My goal for the demo on Tuesday is to get JS devs interested in IoT with Arduino and such. Just trying to wet their appetite. Since wireless is not working out for me currently, I will probably scale back to running over USB which I've done extensively and will have no issue with. But I am interested to continue debugging this with you, even if it means finding out the legacy shield won't work at all!

soundanalogous, 2017-05-04:
I don't see any obvious issues in the sketch files. Which OS are you using (there have been reports of issues connecting over Ethernet when using Ubuntu 16.04 and similar problems could exist for Wi-Fi)?

soundanalogous, 2017-05-04:
If you can't find a WiFi Shield 101, you could get an MKR1000 or an Adafruit Feather Huzzah ESP8266 as those boards definitely work and are cheaper than the WiFi Shield.

soundanalogous, 2017-05-04:
I also assume you're using WPA since WEP doesn't work.

troywweber7, 2017-05-06:
Yes, I'm using WPA. I am using Ubuntu 17.04 and connecting to a NETgear router. I'm going to look into the port blocking. I hadn't thought of that.

troywweber7, 2017-05-07:
I also tried uploading some wifi-shield examples and one of them said to "please update the shield firmware" so I'm trying that as well...

troywweber7, 2017-05-07:
Okay, so I was unable to even `telnet 192.168.1.7 3030` originally, but after [updating the wifi-shield](https://www.arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading) firmware, I was able to telnet in (which means the port is not blocked).

troywweber7, 2017-05-07:
AHHHHH! IT CONNECTS NOW!

Takes a few seconds, though, that's for sure. I'm hoping this will help out someone else. I'm closing for now.

troywweber7, 2017-05-07:
I lied! I was able to connect when I upload the standard firmata wifi. It takes a few seconds for it connect to the standard firmware. However, with configurable firmata wifi, I still seem to be having issues. It has been hanging for several minutes, now, however. I can telnet, into it, though.

The debug console claims the tcp is connected, as well:

```
ConfigurableFirmataWiFi will attempt a WiFi connection using the legacy WiFi library.
IP will be requested from DHCP ...
Attempting to connect to WPA SSID: GS7
WiFi setup done
SSID: GS7
IP Address: 192.168.1.7
signal strength (RSSI): -50 dBm
TCP connection disconnected
TCP connection established
```

soundanalogous, 2017-05-07:
Have you tried using [firmatabuilder](http://firmatabuilder.com/) to create a sketch file with a minimal feature set?

troywweber7, 2017-05-07:
When I was at firmata builder I never actually clicked the serial dropdown, so I was working from the original ConfigurableFirmataWifi.ino file. I'll try the one from the firmata builder now.

troywweber7, 2017-05-07:
Alright, I don't know what I did wrong using the straight `ConfigurableFirmataWifi.ino` file, but using the one off the firmatabuilder website seems to be working fine so far.

closed 2017-05-07 by troywweber7