firmata/ConfigurableFirmata #74: timeout on Genuino 101

haaslukas, 2017-11-23:
I'm working with the Johnny Five client library on the new UDOO X86, which also has a Genuino 101 on board. The blink example worked perfectly with the StandardFirmataPlus sketch. Now, I wanted to add some stepper functionality and, therefore, changed to the ConfigurableFirmata sketch I've download via the firmatabuilder.com (see below).

Unfortunately, I can't even get the blink example to work. I'm always getting a timeout:
`1511431805473 Available /dev/ttyACM0  `
`1511431805509 Connected /dev/ttyACM0  `
`1511431815519 Device or Firmware Error A timeout occurred while connecting to the Board. `

`Please check that you've properly flashed the board with the correct firmware.`
`See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting`

`If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  `
`events.js:146`
`      throw err;`
`      ^`

`Error: Uncaught, unspecified "error" event. ([object Object])`
`    at Board.emit (events.js:144:17)`
`    at Board.log (/home/username/Projekte/node_modules/johnny-five/lib/board.js:635:8)`
`    at Board.(anonymous function) [as error] (/home/username/Projekte/node_modules/johnny-five/lib/board.js:646:14)`
`    at Board.<anonymous> (/home/username/Projekte/node_modules/johnny-five/lib/board.js:390:14)`
`    at Timer.listOnTimeout (timers.js:92:15)`

I tried resetting the board. I also reporgrammed it. Nowthing seems to help. Switching back to the StandardFirmataPlux example sketch works, but then again, I can't use the stepper functionality. So, has enyone an idea? Or is the Genuino 101 just not supported yet?

here's the firmatabuilder.com code:
`/*`
` * firmata-test.ino generated by FirmataBuilder`
 ` */`

`#include <ConfigurableFirmata.h>`

`#include <DigitalInputFirmata.h>`
`DigitalInputFirmata digitalInput;`

`#include <DigitalOutputFirmata.h>`
`DigitalOutputFirmata digitalOutput;`

`#include <StepperFirmata.h>`
`StepperFirmata stepper;`

`#include <FirmataExt.h>`
`FirmataExt firmataExt;`

`void systemResetCallback()`
`{`
`  for (byte i = 0; i < TOTAL_PINS; i++) {`
`    if (IS_PIN_ANALOG(i)) {`
`    } else if (IS_PIN_DIGITAL(i)) {`
`      Firmata.setPinMode(i, OUTPUT);`
`    }`
`  }`
`  firmataExt.reset();`
`}`

`void initTransport()`
`{`
`  // Uncomment to save a couple of seconds by disabling the startup blink sequence.`
`  // Firmata.disableBlinkVersion();`
`  Firmata.begin(57600);`
`}`

`void initFirmata()`
`{`
`  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);`

`  firmataExt.addFeature(digitalInput);`
`  firmataExt.addFeature(digitalOutput);`
`  firmataExt.addFeature(stepper);`
`  Firmata.attach(SYSTEM_RESET, systemResetCallback);`
`}`

`void setup()`
`{`
`  initFirmata();`
`  initTransport();`
`  systemResetCallback();`
`}`

`void loop()`
`{`
`  digitalInput.report();`
`  while(Firmata.available()) {`
`    Firmata.processInput();`
`  }`

`  stepper.update();`
`}`

soundanalogous, 2017-11-23:
I'll look into it. Which pins do you have the stepper motor wired up to?

soundanalogous, 2017-11-23:
Also, which type of stepper driver are you using? Do you have a step + direction driver like the EasyDriver or are you using a 2 or 4 wire h-bridge configuration?

soundanalogous, 2017-11-23:
Ok, figured it out. You need to include AnalogInputFirmata and AnalogOutputFirmata when selecting features in FirmataBuilder. This is because johnny-five expects the analog input and output features and doesn't throw a specific error when they are not included.

closed 2017-11-23 by soundanalogous

soundanalogous, 2017-11-23:
The actual failure is either in firmata.js or johnny-five.js or both. I need to look into that further.

haaslukas, 2017-11-23:
thanks for your quick reply. just in case you need the information for looking into it: I've tried several pins (but I thinks that's not the issue here) and I'm using the EasyDriver v4.4

rwaldron, 2017-11-27:
@haaslukas can you try this version:

```ino
/*
 * customfirmata.ino generated by FirmataBuilder
 * Mon Nov 27 2017 16:20:02 GMT-0500 (EST)
 */

#include <ConfigurableFirmata.h>

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;

#include <Servo.h>
#include <ServoFirmata.h>
ServoFirmata servo;

#include <Wire.h>
#include <I2CFirmata.h>
I2CFirmata i2c;

#include <StepperFirmata.h>
StepperFirmata stepper;

#include <FirmataExt.h>
FirmataExt firmataExt;

#include <AnalogWrite.h>

#include <FirmataReporting.h>
FirmataReporting reporting;

void systemResetCallback()
{
  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_ANALOG(i)) {
      Firmata.setPinMode(i, ANALOG);
    } else if (IS_PIN_DIGITAL(i)) {
      Firmata.setPinMode(i, OUTPUT);
    }
  }
  firmataExt.reset();
}

void initTransport()
{
  // Uncomment to save a couple of seconds by disabling the startup blink sequence.
  // Firmata.disableBlinkVersion();
  Firmata.begin(57600);
}

void initFirmata()
{
  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(servo);
  firmataExt.addFeature(i2c);
  firmataExt.addFeature(stepper);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);
}

void setup()
{
  initFirmata();

  initTransport();

  systemResetCallback();
}

void loop()
{
  digitalInput.report();

  while(Firmata.available()) {
    Firmata.processInput();
  }

  if (reporting.elapsed()) {
    analogInput.report();
    i2c.report();
  }

  stepper.update();
}
```