firmata/ConfigurableFirmata #78: AccelStepper custom float format decoding

dylanbannon, 2018-01-08:
Looking at the function AccelStepperFirmata::decode32BitSignedInteger, it doesn't look like the expected encoding of the float is the same as outlined in the Firmata protocol:
https://github.com/firmata/protocol/blob/master/accelStepperFirmata.md#accelstepperfirmatas-custom-float-format

Idk whether this is an issue with ConfigurableFirmata or the protocol specification, but it can be confusing for newcomers.

soundanalogous, 2018-01-08:
@dtex can you take a look?

dtex, 2018-01-08:
Hi @dylanbannon That function is for 32-bit signed integers while that section of the protocol documentation is for the custom float format. Two different number types.

dylanbannon, 2018-01-08:
Whoops, sorry, @dtex . I included the wrong signature in my comment. I should have put AccelStepperFirmata::decodeCustomFloat.

I think my only issue is just the diagram in the protocol documentation. It seems to be completely different from the implementation here, but I think I could be misreading it.

As an example of what I'm talking about, the diagram seems to indicate that the first 21 bits of the significand will be transmitted as the least significant bits of the message, and then the last 2 bits of the significand will be transmitted as the two most significant bits, with the sign and exponent stuffed in between those two sets of bits.

The implementation in AccelStepperFirmata::decodeCustomFloat definitely doesn't follow that scheme and, tbh, the approach that's in the code seems more logical than what's described in the diagram.

dtex, 2018-01-08:
I found a small error, but It sounds like you're seeing a larger disparity. Can you elaborate on what you are seeing? I could only find a 1 bit error in the significand for example 1. 

|            | Decimal| Binary                 |
|------------|--------|------------------------|
|Significand | 2777777| 01010100110001010110010|
|Exponent    |       1|                    0001|
|Sign        |       0|                       0|

Should actually be

|            | Decimal| Binary                 |
|------------|--------|------------------------|
|Significand | 2777777| 01010100110001010110001|
|Exponent    |       1|                    0001|
|Sign        |       0|                       0|

I could add a column to the 3rd table in each example saying which byte each row represents. That might add a little clarity:

|            |                                           | Binary  | Hex |
|-------|-------------------------------------------|---------|-----|
| Byte 1 | Least most significant bits               |  0110001| 0x31|
| Byte 2 | Next most significant bits                |  1000101| 0x45|
| Byte 3 | Next most significant bits                |  0101001| 0x29|
| Byte 4 | Sign, Exponent and 2 most significant bits|  0000101| 0x05|

dylanbannon, 2018-01-08:
The part I'm talking about is this diagram, just above the examples:

[image]

Am I just interpreting it incorrectly?

dtex, 2018-01-09:
It's right, I just think it is confusing. Should probably break 0-20 into three groups: Least significant 7-bits, next most significant, next most significant. It's confusing because the bytes increase in significance as you go left to right but within each byte the bits decrease in significance as you go left to right.

dylanbannon, 2018-01-09:
Ah, I understand it now. Thanks for clarifying that. Looks like I was coding a little beyond my depth and got confused.

dtex, 2018-01-09:
I appreciate you looking closely at it. 

I'm pretty sure the documentation could be better. For now I'm going fix that error on example 1.

dtex, 2018-01-09:
The bit error was fixed by https://github.com/firmata/protocol/pull/100 so I think we can close this.

closed 2018-06-02 by soundanalogous