firmata/ConfigurableFirmata #79: NodeMCU keeps opening Hotspot in Client Mode

relief-melone, 2018-04-28:
Hey guys,

I'm really having trouble connecting to my ESP8266 today. I was working with this setup for  long time now and I didn't have to flash my boards. Now after doing the following

- Updating to the newest ConfigurableFirmate
- Using FirmataBuilder to download firmata
- Uploading Firmata

I can't connect via Johnny-Five anymore (getting the usual timeout error). I couldn't narrow down what it exaclty is (and I have been trying all day) but these are things that I have noticed (Using ConfigurableFirmata directly as well as using firmatabuilder)

- an open Hotspot gets created even though I'm in Client Mode
- when scanning for ports I do find the Module connected to my Wifi (Serial Monitor in Debug tells me the same too) but no open port is found

I posted both of these issues here, as i don't know if they are connected. I also opened an issue in the johnny-five-repo because I couldn't narrow down the reason. I hope you guys can.

soundanalogous, 2018-04-29:
There have not been any changes to the WiFi functionality in ConfigurableFirmata in several months so I doubt this issue has anything to do with ConfigurableFirmata. However, I'll take a look with my ESP8266 board in the next few days to see if I can repro.

relief-melone, 2018-05-08:
OK,

this was a simple user mistake. But since I lost a whole day searching for the error and I think others might too, I'll share my mistake here.

As I pointed out I created the Firmata as Client (thinking its a client to my Hotspot). But you will have to create it as a server (as you want to connect to the board)

It's as simple as that. If you are on http://firmatabuilder.com/ make sure you choose **TCP Server** if you want to connect to your board.

closed 2018-05-08 by relief-melone