firmata/ConfigurableFirmata #81: setup call to systemResetCallback causes undesirable analog reporting behavior

ethanjli, 2018-05-28:
# Problem
Currently, in the example sketch (and in sketches generated by Firmata Builder), `systemResetCallback()` is called at the end of `setup()` (as in [line 377 of the example ConfigurableFirmata.ino](https://github.com/firmata/ConfigurableFirmata/blob/2.10.0/examples/ConfigurableFirmata/ConfigurableFirmata.ino#L377) and in [line 316 of firmata-builder's builder.js](https://github.com/firmata/firmata-builder/blob/v0.6.1/lib/builder.js#L316)). However, this call does not trigger the other work done in the private `FirmataClass::systemReset()` method, most notably adjustment of the private `resetting` member and its `isResetting()` getter.

The problematic consequence is that, when `systemResetCallback` is called from `setup()`, it will cause `AnalogInputFirmata::reportAnalog(...)` to spuriously make a `Firmata.sendAnalog(analogPin, analogRead(analogPin));` call within the `if (!Firmata.isResetting())` branch. I believe this is the same issue which was fixed in [issue 185 of firmata/arduino](https://github.com/firmata/arduino/issues/185). I first identified this issue when troubleshooting the same problem which was described there, except that I am using a ConfigurableFirmata sketch from Firmata Builder rather than StandardFirmata (see below).

# Proposed Solution
I believe that this issue can be solved by replacing the `systemResetCallback()` call at the end of `setup()` in `ConfigurableFirmata.ino` with either:
1. a direct call to `FirmataClass::systemReset()` (which would require making that private method public), or
2. a call to `FirmataClass::parse(...)` with the appropriate `SYSTEM_RESET` as the argument, which has the same effect as just calling `FirmataClass::systemReset()` directly, at least when this call is made at the end of `setup()`.

I've confirmed with basic testing (see below) that option 2 seems to solve the problematic consequence described above, and it does not require making the private `FirmataClass::systemReset()` method public, simply a one-line change to `examples/ConfigurableFirmata/ConfigurableFirmata.ino` and a one-line change to [firmata/firmata-builder](https://github.com/firmata/firmata-builder)'s `lib/builder.js`. Would this be a reasonable fix?

# Reproducing the problem
## Testing Conditions
I installed [pymata-aio 2.20](https://pypi.org/project/pymata-aio/), connected an Arduino Uno running the ConfigurableFirmata sketch, and attempted to run a [blink script](https://github.com/MrYsLab/pymata-aio/blob/master/examples/blink.py). With the problematic sketch below, pymata receives an analog report of a pin before it receives an analog mapping response, which causes a crash of the script (to be specific, pymata attempts to index with a pin number into an empty list of pins, leading to an IndexError and graceful exit). With the proposed fix below, the blink script runs normally.

## Problematic Sketch
The following sketch does not work for me:
```c++
#include <ConfigurableFirmata.h>

#include <DigitalInputFirmata.h>
#include <DigitalOutputFirmata.h>
#include <AnalogInputFirmata.h>
#include <AnalogOutputFirmata.h>
#include <AnalogWrite.h>
#include <FirmataReporting.h>
#include <FirmataExt.h>

DigitalInputFirmata digitalInput;
DigitalOutputFirmata digitalOutput;
AnalogInputFirmata analogInput;
AnalogOutputFirmata analogOutput;

FirmataReporting reporting;

FirmataExt firmataExt;

void initTransport()
{
  // Uncomment to save a couple of seconds by disabling the startup blink sequence.
  Firmata.disableBlinkVersion();
  Firmata.begin(57600);
}

void initFirmata()
{
  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);
}

void setup()
{
  initFirmata();

  initTransport();

  // ------------------------
  // The problematic line
  // ------------------------
  systemResetCallback();
}

void loop() {
  digitalInput.report();

  while (Firmata.available()) Firmata.processInput();

  if (reporting.elapsed()) analogInput.report();
}
```

## Proposed Fix
The following sketch works for me:
```c++
#include <ConfigurableFirmata.h>

#include <DigitalInputFirmata.h>
#include <DigitalOutputFirmata.h>
#include <AnalogInputFirmata.h>
#include <AnalogOutputFirmata.h>
#include <AnalogWrite.h>
#include <FirmataReporting.h>
#include <FirmataExt.h>

DigitalInputFirmata digitalInput;
DigitalOutputFirmata digitalOutput;
AnalogInputFirmata analogInput;
AnalogOutputFirmata analogOutput;

FirmataReporting reporting;

FirmataExt firmataExt;

void systemResetCallback()
{
  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_ANALOG(i)) {
      Firmata.setPinMode(i, ANALOG);
    } else if (IS_PIN_DIGITAL(i)) {
      Firmata.setPinMode(i, OUTPUT);
    }
  }
  firmataExt.reset();
}

void initTransport()
{
  // Uncomment to save a couple of seconds by disabling the startup blink sequence.
  Firmata.disableBlinkVersion();
  Firmata.begin(57600);
}

void initFirmata()
{
  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);
}

void setup()
{
  initFirmata();

  initTransport();

  // ------------------------
  // Proposed fix #2
  // ------------------------
  Firmata.parse(SYSTEM_RESET); 
}

void loop() {
  digitalInput.report();

  while (Firmata.available()) Firmata.processInput();

  if (reporting.elapsed()) analogInput.report();
}
```

soundanalogous, 2018-05-28:
Calling `Firmata.parse(SYSTEM_RESET);` feels like a hack to me, however this change would work in both ConfigurableFirmata and Firmata at the sketch level without changing any of the classes. Ideally, I'd rather make `Firmata::systemReset` public, but that would require additional changes to firmata/arduino since `Firmata::systemReset` was moved to `FirmataParser::systemReset` and I don't want to have to call methods on `FirmataParser` directly from a sketch.

soundanalogous, 2018-05-28:
@ethanjli do you want to submit PRs with the necessary changes to ConfigurableFirmata and firmata-builder? You may need to update the unit tests in firmata-builder so if you want to take on the PR, please make sure you run the tests.

closed 2018-06-02 by soundanalogous