firmata/ConfigurableFirmata #85: firmdata client on ready never fire?

wyx2000, 2018-11-06:
I started with RCSwitchFirmata, could not get it work.  I think I should test ConfigurableFirmata first. So I upload ConfigurableFirmata to my UNO, and use a client code with firmata lib like below:
`var firmata = require('firmata');
var board = new firmata.Board('COM4', function () {
  console.log('ready!')
});
 
  board.on("ready", function() {
    const a = 6;
    const b = 7;
.....
`
neither ready above fires. But if I try it with the StandardFirmata code, it works fine.

Any idea? Is there special client side code for ConfigurableFirmata ?

Thanks

soundanalogous, 2018-11-07:
Which classes did you include in your ConfigurableFirmata sketch file? If you include too many it's possible there is not enough memory on the Uno and the program will crash silently.

git-developer, 2018-11-10:
This issue has probably been solved in [RCSwitchFirmata #2](https://github.com/git-developer/RCSwitchFirmata/issues/2) and [RCSwitchFirmata #3](https://github.com/git-developer/RCSwitchFirmata/issues/3).