firmata/ConfigurableFirmata #93: Using ConfigurableFirmataWiFi with thermometer DS18B20 and Johnny-Five not working

juliagf, 2020-04-05:
Hi! I'm trying to make work a DS1820 thermometer with ConfigurableFirmataWiFi firmware flashed on my NodeMCU 1.0 Amica ESP-12E board, but it does not work. Steps to reproduce:

- NodeMCU 1.0 Amica ESP-12E flashed with ConfigurableFirmataWiFi, connected to my home WiFi. It seems to work as **it is working perfectly with an infrared proximity sensor.** **Temperature sensor seems to be correctly connected as is working with another script** which shows the temperature every 3 seconds via WiFi on serial monitor.

- My code on the server script:
[image]
- The output:
>>1586111990178 SerialPort Connecting to host:port: 192.168.0.22:3030  
1586111990180 Connected Connecting to host:port: 192.168.0.22:3030  
1586111995233 Repl Initialized  
BOARD 2 READY!
<ref *1> Thermometer {
  board: Board {
    port: 'Connecting to host:port: 192.168.0.22:3030',
    timer: Timeout {
      _idleTimeout: -1,
      _idlePrev: null,
      _idleNext: null,
      _idleStart: 166,
      _onTimeout: null,
      _timerArgs: undefined,
      _repeat: null,
      _destroyed: true,
      [Symbol(refed)]: true,
      [Symbol(asyncId)]: 12,
      [Symbol(triggerId)]: 1
    },
    isConnected: true,
    isReady: true,
    io: Firmata {
      _events: [Object: null prototype],
      _eventsCount: 5,
      _maxListeners: undefined,
      isReady: true,
      MODES: [Object],
      I2C_MODES: [Object],
      STEPPER: [Object],
      SERIAL_MODES: [Object],
      SERIAL_PORT_IDs: [Object],
      SERIAL_PIN_TYPES: [Object],
      RESOLUTION: [Object],
      HIGH: 1,
      LOW: 0,
      pins: [Array],
      ports: [Array],
      analogPins: [Array],
      version: [Object],
      firmware: [Object],
      buffer: [],
      versionReceived: true,
      name: 'Firmata',
      settings: [Object],
      pending: 2,
      digitalPortQueue: 0,
      transport: [EtherPortClient],
      reportVersionTimeoutId: Timeout {
        _idleTimeout: 5000,
        _idlePrev: null,
        _idleNext: null,
        _idleStart: 165,
        _onTimeout: null,
        _timerArgs: undefined,
        _repeat: null,
        _destroyed: true,
        [Symbol(refed)]: true,
        [Symbol(asyncId)]: 11,
        [Symbol(triggerId)]: 1
      },
      defaultLed: 13,
      port: 'Connecting to host:port: 192.168.0.22:3030',
      [Symbol(kCapture)]: false
    },
    register: [ [Circular *1] ],
    occupied: [],
    Drivers: {},
    id: 'B9952B08-A394-44F1-9FE1-96E746A50E4E',
    debug: true,
    repl: Repl { context: [Object], ready: true, cmd: [REPLServer] },
    sigint: true,
    pins: Pins {
      '0': [Object],
      '1': [Object],
      '2': [Object],
      '3': [Object],
      '4': [Object],
      '5': [Object],
      '6': [Object],
      '7': [Object],
      '8': [Object],
      '9': [Object],
      '10': [Object],
      '11': [Object],
      '12': [Object],
      '13': [Object],
      '14': [Object],
      '15': [Object],
      '16': [Object],
      '17': [Object]
    },
    _events: [Object: null prototype] { ready: [Function (anonymous)] },
    _eventsCount: 1,
    type: 'OTHER',
    MODES: {
      INPUT: 0,
      OUTPUT: 1,
      ANALOG: 2,
      PWM: 3,
      SERVO: 4,
      SHIFT: 5,
      I2C: 6,
      ONEWIRE: 7,
      STEPPER: 8,
      SERIAL: 10,
      PULLUP: 11,
      IGNORE: 127,
      PING_READ: 117,
      UNKOWN: 16
    },
    RESOLUTION: { ADC: 1023, DAC: null, PWM: 1023 },
    millis: [Function (anonymous)]
  },
  io: Firmata {
    _events: [Object: null prototype] {
      close: [Function: bound ],
      disconnect: [Function: bound ],
      error: [Function: bound ],
      string: [Function: bound ],
      '1-wire-search-reply-1': [Function]
    },
    _eventsCount: 5,
    _maxListeners: undefined,
    isReady: true,
    MODES: {
      INPUT: 0,
      OUTPUT: 1,
      ANALOG: 2,
      PWM: 3,
      SERVO: 4,
      SHIFT: 5,
      I2C: 6,
      ONEWIRE: 7,
      STEPPER: 8,
      SERIAL: 10,
      PULLUP: 11,
      IGNORE: 127,
      PING_READ: 117,
      UNKOWN: 16
    },
    I2C_MODES: { WRITE: 0, READ: 1, CONTINUOUS_READ: 2, STOP_READING: 3 },
    STEPPER: {
      TYPE: [Object],
      STEP_SIZE: [Object],
      RUN_STATE: [Object],
      DIRECTION: [Object]
    },
    SERIAL_MODES: { CONTINUOUS_READ: 0, STOP_READING: 1 },
    SERIAL_PORT_IDs: {
      HW_SERIAL0: 0,
      HW_SERIAL1: 1,
      HW_SERIAL2: 2,
      HW_SERIAL3: 3,
      SW_SERIAL0: 8,
      SW_SERIAL1: 9,
      SW_SERIAL2: 16,
      SW_SERIAL3: 17,
      DEFAULT: 8
    },
    SERIAL_PIN_TYPES: {
      RES_RX0: 0,
      RES_TX0: 1,
      RES_RX1: 2,
      RES_TX1: 3,
      RES_RX2: 4,
      RES_TX2: 5,
      RES_RX3: 6,
      RES_TX3: 7
    },
    RESOLUTION: { ADC: 1023, DAC: null, PWM: 1023 },
    HIGH: 1,
    LOW: 0,
    pins: [
      [Object], [Object], [Object],
      [Object], [Object], [Object],
      [Object], [Object], [Object],
      [Object], [Object], [Object],
      [Object], [Object], [Object],
      [Object], [Object], [Object]
    ],
    ports: [
      0, 0, 0, 0, 0, 0,
      0, 0, 0, 0, 0, 0,
      0, 0, 0, 0
    ],
    analogPins: [ 17 ],
    version: { major: 2, minor: 6 },
    firmware: { name: 'ConfigurableFirmataWiFi.ino', version: [Object] },
    buffer: [],
    versionReceived: true,
    name: 'Firmata',
    settings: {
      reportVersionTimeout: 5000,
      samplingInterval: 19,
      serialport: [Object]
    },
    pending: 2,
    digitalPortQueue: 0,
    transport: EtherPortClient {
      _events: [Object: null prototype],
      _eventsCount: 4,
      _maxListeners: undefined,
      path: 'Connecting to host:port: 192.168.0.22:3030',
      name: 'EtherPortClient',
      host: '192.168.0.22',
      port: 3030,
      _reconnectTimeoutSecs: 15,
      _queue: [],
      _socket: [Socket],
      _reconnectTimer: null,
      _tcp: [Socket],
      [Symbol(kCapture)]: false
    },
    reportVersionTimeoutId: Timeout {
      _idleTimeout: 5000,
      _idlePrev: null,
      _idleNext: null,
      _idleStart: 165,
      _onTimeout: null,
      _timerArgs: undefined,
      _repeat: null,
      _destroyed: true,
      [Symbol(refed)]: true,
      [Symbol(asyncId)]: 11,
      [Symbol(triggerId)]: 1
    },
    defaultLed: 13,
    port: 'Connecting to host:port: 192.168.0.22:3030',
    [Symbol(kCapture)]: false
  },
  id: '443E2C98-9F7F-4221-AF85-4D11549033CD',
  custom: {},
  pin: 1,
  controller: 'DS18B20',
  aref: 5
}
events.js:298
      throw er; // Unhandled 'error' event
Error: 1-Wire device search timeout - are you running ConfigurableFirmata?
    at Timeout._onTimeout (/Users/juliagf/dev/control-de-riego-backend/node_modules/firmata-io/lib/firmata.js:1523:16)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Emitted 'error' event at:
    at EventEmitter.<anonymous> (/Users/juliagf/dev/control-de-riego-backend/node_modules/johnny-five/lib/thermometer.js:185:18)
    at Timeout._onTimeout (/Users/juliagf/dev/control-de-riego-backend/node_modules/firmata-io/lib/firmata.js:1523:7)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

I tried to change the "pin" field to: 'D1', 1, 5. Also with the "change" event. Nothing worked for me. I tried to reinstall ConfigurableFirmata. Nothing. I tried to build the firmware with the FirmataBuilder. Nothing. I saw there is another issue #1347 that was closed without solving. I don't know what else I could do. Totally desperate. Some help would be great. Thanks so much in advance.

soundanalogous, 2020-04-08:
I've encountered issues using a DS18B20 when other sensors are connected to the board. Never been able to determine the exact root of this issue. Best route is to use an I2C or a simple analog based thermometer and avoid OneWire with Firmata.

juliagf, 2020-04-09:
> I've encountered issues using a DS18B20 when other sensors are connected to the board. Never been able to determine the exact root of this issue. Best route is to use an I2C or a simple analog based thermometer and avoid OneWire with Firmata.

Thank you for your quick answer and honesty, I will try to use another thermometer and won't spend more time in this...  👍

closed 2020-04-09 by juliagf