firmata/ConfigurableFirmata #100: Default Low/High on startup

PizzaProgram, 2020-11-21:
I think it would make life easier for beginner {like me ;-) } if the [firmatabuilder](http://firmatabuilder.com/) would "ask about" the default settings of output pins.  
  
Temporary I'll try to use this method mentioned here:  #https://github.com/firmata/arduino/issues/166#issuecomment-330331475  
but it can have serious consequences if predefined to a different state as it should be when the board starts.  
  
< OFF >  
_For example:  
 I'm building currently a 8+8 relay-board [roller-shutter](https://www.google.com/search?q=roller+shutter+metal+door) system.  
Those 230V tube-motors are very strong and those heavy metallic things can basically cut off a child's head..._  
< /OFF >

soundanalogous, 2020-11-24:
It's more complicated that that. The ability to [enable reporting of individual digital pins](https://github.com/firmata/protocol/issues/68) needs to be added first, which is a big change from how Firmata has historically worked so it will not land until Firmata 3.0, which has no definite schedule.

soundanalogous, 2020-11-24:
My advice is if you are controlling potentially dangerous equipment, is not to use Firmata, but to design a simple protocol that will meet your needs while giving you full control over the code running on your Arduino.

PizzaProgram, 2020-12-01:
I understand.
The good news is: it seems the code works.
```
void setup()
{
# Uncomment following 2 lines, if you want to start the board on power-up at LOW or HIGH state
#  byte pin;
#  for (pin = 0; pin < TOTAL_PINS; pin++) {     digitalWrite(pin, HIGH);     }
...
```

I recommend to copy this few lines to :
StandardFirmata.ino (ca. at line 700+)

And firmatabuilder should have an extra checkbox to enable it too.

That would be already a HUGE help to us, beginners.

< OFF >
IMHO You can not prohibit people to use your code at automatisation systems.
Especially because Node-Red even recommends to use it, and there are some industrial systems done with it already.

My personal opinion is, that currently Firmata is more stable than any other "ad-hock" created serial service.
 (Maybe [Ardulink](http://www.ardulink.org) has still some chance to grow to IoT standard, but Firmata already **IS one**.)
< /OFF >

PizzaProgram, 2024-02-22:
Is it possible to define a global variable (or activate any hidden feature) to:  
## force all pins to work in INVERTED mode?  
Where `LOW` becomes `HIGH` in all cases, and the other way around. Also all output pins are initialized `HIGH` at the beginning instead of `LOW`.  
  
_Something like:_  
```c
#define ENABLE_INVERT_MODE 
```

pgrawehr, 2024-02-25:
No, there's currently no such option. However,  the startup values are directly in the ino file and can easily be changed there.

closed 2024-02-25 by PizzaProgram

PizzaProgram, 2024-02-25:
Rather than waiting, 
created a pull req. myself.
https://github.com/firmata/ConfigurableFirmata/pull/162