firmata/ConfigurableFirmata #114: Firmata with HA

tomasz-grobelny, 2021-07-17:
Is there any good tutorial how to connect Arduino to Home Assistant using Firmata? To be more specific I was thinking that I could use Firmata to have the following functionality:
1. Arduino Uno or similar connected to monostable wall switch and relay.
2. If I press the switch the relay toggles (eg. on falling edge) and HA gets notified somehow of the new state.
3. If I long press the switch HA gets notified that long press happened and I can run an automation.

Is this possible with Firmata? Any documentation on similar use case?

pgrawehr, 2021-07-24:
What kind of "Home Assistant" are you talking about (there's a bunch of such devices available)? What kind of interfaces (Network, USB, etc) does it have? Can you program it?

tomasz-grobelny, 2021-07-27:
https://www.home-assistant.io/integrations/firmata/ - this the one I am talking about. It seems it can communicate using serial/serial over USB.

Alternative way would be to use mqtt integration, but then: are you aware of any firmata to mqtt bridge?

pgrawehr, 2021-07-27:
@tomasz-grobelny The link you provided seems to answer your question. Yes, firmata can be used to control lights and other devices from HA and also recognize switches. What you need to do is described there.

pgrawehr, 2022-01-20:
Closing, seems solved by the provided link.

closed 2022-01-20 by pgrawehr