firmata/ConfigurableFirmata #119: Frequency pin mode throwing "Unknown pin mode" in FirmataExt

eric-williams-design, 2021-07-28:
I know this is the newest addition to the ConfigurableFirmata groups but I'm having issue getting reliable responses and I see the error is being thrown within FirmataExt.cpp. Perhaps I didn't configure firmata appropriately or my client code has some issue, but I wonder if you could help with the documentation for implementing client support.

pgrawehr, 2021-08-02:
Can you write a bit more precisely what you did and where it failed? What client library where you using?

Frequency pin mode is new, and maybe your client code doesn't handle an unknown pin mode gracefully.

eric-williams-design, 2021-08-02:
Thanks for the reply. I have been working on updating Firmata for Processing so the client side is my code. I believe the error I am having is coming from setting the pin mode to FREQUENCY within the client. If I simply setup the pin to be INPUT and not FREQUENCY the error goes away. I'm up against several sources of error because I am somewhat new to JAVA, but I am getting it to work. Do you have any additional notes on how to use your package or how you might extend it to monitor two or more pins for frequency measurement? I think I follow the basics...

pgrawehr, 2021-08-02:
Nice someone has an use for the new feature ;-)

You are right that something with the pin mode looks inconsistent. I'll check that. I implemented the feature for one pin only just to save memory. When multiple pins should be supported, an array of pin numbers is required, which quickly uses quite some memory on the AVR-based Arduinos. But it shouldn't be <i>difficult</i> to implement it.

eric-williams-design, 2021-08-02:
I saw the tool post as I was looking at implementing frequency measurement myself if it wasn't available. I've been only following the firmata protocol for a couple months now, but I'm happy to see new features! I have a use case where I'm working with a CNC machine with a variety of sensors and need to advance the interrupts a bit more too. If you have any advice for adding additional interrupts that would be great.

EXTERNAL:

You are right that something with the pin mode looks inconsistent. I'll check that. I implemented the feature for one pin only just to save memory. When multiple pins should be supported, an array of pin numbers is required, which quickly uses quite some memory on the AVR-based Arduinos. But it shouldn't be difficult to implement it.

-

pgrawehr, 2021-08-02:
The code is in Frequency.cpp/Frequency.h. You'll have to keep arrays of some variables instead of just one. I may take a look at it later, but I'm traveling right now, so I can't say when I'll have time. Note also that the Arduino UNO has only two pins that are interrupt-capable.

pgrawehr, 2021-08-03:
I have fixed a small inconsistency (the mode was not properly set, depending on the command sequence used). I'm not sure this caused your issue in the client, though.

pgrawehr, 2021-09-20:
@eric-williams-design Is there something to do here still, or can I close this issue?

closed 2021-09-20 by eric-williams-design