firmata/ConfigurableFirmata #120: Lost compatibility with Arduino Boards update 1.8.3

eric-williams-design, 2021-08-10:
I have been working with the Configurable Firmata source code for a while now and something just crashed with the new boards update. It seems to be rejecting something in the serial commands for the Arduino Mega 2560. I'm not sure if this is just my build or not, but I went back to the previous board manager release 1.8.2 and things worked fine. 
[boards manager update error.txt](https://github.com/firmata/ConfigurableFirmata/files/6963431/boards.manager.update.error.txt)

pgrawehr, 2021-08-11:
Looks like a problem with your installation or with that update. The ConfiguarableFirmata version you're using seems to be the official release, which hasn't changed. It doesn't find anything,  actually. Maybe it needs a rebuild.

eric-williams-design, 2021-08-11:
I just rebuilt and it works fine now! thanks!

closed 2021-08-11 by eric-williams-design