firmata/ConfigurableFirmata #131: Using MKR1010 WiFi

hgmarques, 2022-10-14:
Hello...

I'm having trouble running ConfigurableFirmata.ino with the board MKR1010 Wifi. Is it actually possible? The problem lies with using the appropriate WiFi library. Which one  should I use?

Thanks in advance.

Regards,
hugo

pgrawehr, 2022-10-15:
I tested wifi only with ESP32, therefore I'm not sure,  but I think it used to work. Can you give more information about the errors you get? Does it fail during build or only at runtime?

mphasize, 2022-12-22:
Hi, I'm currently trying to do the same, getting ConfigurableFirmata to run on the Arduino MKR WiFi 1010. 

As a step 1, I have to edit the `Boards.h` file to add the 1010. Fortunately there is already an entry in the original Firmata repository here https://github.com/firmata/arduino/blob/e8781a40e8d94597d7783b59f368fb5ce85f67a4/Boards.h#L313

As a step 2, we need to use the WiFiNINA library instead of WiFi101 as that is the one compatible with the MKR WIFI 1010.

Now we get a compile error, as `SET_PIN_MODE` is defined in the global namespace in both the WiFiNINA library and ConfigurableFirmata. The same issue is described here in the Firmata repo: https://github.com/firmata/arduino/issues/470

For the regular Firmata there is a suggested fix, as `SET_PIN_MODE` is actually set to the namespaced `firmata::SET_PIN_MODE` which is used internally. 

Unfortunately I cannot try the same workaround for ConfigurableFirmata as the definition https://github.com/firmata/ConfigurableFirmata/blob/master/src/ConfigurableFirmata.h#L56 seems to indicate that ConfigurableFirmata is not using namespaces and we cannot avoid the clashing definitions. 

This is as far as I got. Any ideas for a potential solution or workaround would be highly appreciated. :)

pgrawehr, 2022-12-23:
@mphasize Hi Marcus, thanks for reporting this.

I don't have such a board, so I'll just be able to assist in getting it to build, you'll need to test it yourself. Do you by chance already create a branch on a fork I can start out with?

mphasize, 2022-12-23:
Hi, thanks for the offer. For now, I managed to do a workaround by renaming the `SET_PIN_MODE` constant in the WiFiNINA library and leaving ConfigurableFirmata as it is. I would assume that going forward it might make sense to refactor the library to work with namespaces, but unfortunately I have very little experience with C/C++.

I hope I will find some time to create a PR to add at least the Board definition and maybe a conditional import of the WiFiNINA as a replacement for WiFi101. 

I also opened an issue here https://github.com/arduino-libraries/WiFiNINA/issues/233 so that the clashing globals are at least documented.