firmata/ConfigurableFirmata #139: esp32 not work, still timeout on "ready" connection.

awong1900, 2023-03-30:
I use esp32 with ConfigurableFirmata 3.0.0. The firmware use `ConfigurableFirmata.ino`.  Success flash to esp32, and output
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
⸮⸮qB o o t i n g   d e v i c e .   S t a n d   b y . . . ⸮E (1161) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1161) gpio: GPIO can only be used as input mode
E (1161) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1165) gpio: GPIO can only be used as input mode
```

And i test it with J5 and Firmata.js, like this
```
const { Board, Led } = require("johnny-five");
const board = new Board({ port: "/dev/tty.wchusbserial14230" });

board.on("ready", () => {
  // Create an Led on pin 13
  const led = new Led(13);
  // Blink every half second
  led.blink(500);
});
```
Output:
```
1680186605530 Connected /dev/tty.wchusbserial14230  
1680186615534 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
events.js:305
    throw err; // Unhandled 'error' event
    ^

Error [ERR_UNHANDLED_ERROR]: Unhandled error. ({
  type: 'error',
  timestamp: 1680186615534,
  class: 'Device or Firmware Error',
  message: 'A timeout occurred while connecting to the Board. \n' +
    '\n' +
    "Please check that you've properly flashed the board with the correct firmware.\n" +
    'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
    '\n' +
    "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
  data: null
})
    at Board.emit (events.js:303:17)
    at Board.log (/Users/jump/Documents/Github/mario-usbuart/mario-led-js/node_modules/johnny-five/lib/board.js:637:8)
    at Board.<computed> [as error] (/Users/jump/Documents/Github/mario-usbuart/mario-led-js/node_modules/johnny-five/lib/board.js:648:14)
    at Timeout._onTimeout (/Users/jump/Documents/Github/mario-usbuart/mario-led-js/node_modules/johnny-five/lib/board.js:384:14)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7) {
  code: 'ERR_UNHANDLED_ERROR',
  context: {
    type: 'error',
    timestamp: 1680186615534,
    class: 'Device or Firmware Error',
    message: 'A timeout occurred while connecting to the Board. \n' +
      '\n' +
      "Please check that you've properly flashed the board with the correct firmware.\n" +
      'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
      '\n' +
      "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
    data: null
  }
}

```

Firmata.js code:
```
const Firmata = require("firmata");
const board = new Firmata("/dev/tty.wchusbserial14230");

board.on("open", () => {
  console.log("  ✔ open");
});

board.on("ready", () => {
  // Arduino is ready to communicate
  console.log("ready");
});

var timeout = setTimeout(() => {
  console.log(board.currentBuffer);
  console.log(">>>>>>>>>>>>>>TIMEOUT<<<<<<<<<<<<<<");
  console.log("------------------------------");
  process.exit();
}, 10000);

```
Output:
```
  ✔ open
undefined
>>>>>>>>>>>>>>TIMEOUT<<<<<<<<<<<<<<
------------------------------
```
My esp chip info:
```
$esptool.py -p /dev/tty.wchusbserial14230 flash_id
esptool.py v4.5.1
Serial port /dev/tty.wchusbserial14230
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:b2:9a:b8
Uploading stub...
Running stub...
Stub running...
Manufacturer: 0b
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```
It looks like the ready communication is not complete. Any suggestions would be appreciated.

ale-novo, 2023-04-02:
For me i could not make work J5 for any arduino or arduino compatible board. I dont know why
im using perl-firmata and it works fine, i suggest you give it a try.

https://github.com/ntruchsess/perl-firmata/tree/master/examples

pgrawehr, 2023-04-02:
@ale-novo It seems some change in the latest version has broken the js library. See https://github.com/rwaldron/johnny-five/issues/1817

awong1900, 2023-04-02:
I used UNO board install ConfigurableFirmata 3.0, still not working. It looks like ConfigurableFirmata has issues with j5 or Firmata.js.
I will check perl-firmata follow @ale-novo 's advise.

pgrawehr, 2023-04-04:
@awong1900 Check the baudrate that is used. The default baudrate for ConfigurableFirmata 3.0 is now 115200.

awong1900, 2023-04-04:
Yes, I carefully checked the baud rate.  I think i found true reason on UNO.

Make `initFirmata(); `first, it work on UNO.
```c++
void setup()
{
  initFirmata(); // Change the initialization order.
 
  initTransport();

  Firmata.sendString(F("Booting device. Stand by..."));

  Firmata.parse(SYSTEM_RESET);

  // Firmata.sendString(F("System booted. Free bytes: 0x"), freeMemory());
}
```
Because, Follow the `ConfigurableFirmata.cpp`, we need firstly set the firmware version.
```c++
/**
 * Sends the firmware name and version to the Firmata host application. The major and minor version
 * numbers are the first 2 bytes in the message. The following bytes are the characters of the
 * firmware name.
 */
void FirmataClass::printFirmwareVersion(void)
{
    if (firmwareVersionMajor != 0 && FirmataStream != nullptr) { // make sure that the name has been set before reporting
        startSysex();
        FirmataStream->write(REPORT_FIRMWARE);
        FirmataStream->write(firmwareVersionMajor); // major version number
        FirmataStream->write(firmwareVersionMinor); // minor version number
        size_t len = strlen(firmwareVersionName);
        for (size_t i = 0; i < len; ++i)
        {
            sendValueAsTwo7bitBytes(firmwareVersionName[i]);
        }
        endSysex();
    }
}
```

awong1900, 2023-04-04:
But still not work, it stop on `queryAnalogMapping`.
https://github.com/firmata/firmata.js/blob/4a3b86b46bedf5d238e19ff9aed08742cc607114/packages/firmata-io/lib/firmata.js#L778

ale-novo, 2023-04-04:
ive tested J5 like 1 year ago and i was running ConfigurableFirmata v2.10, and like i said i had no luck for it to connect with several boards. Dont know why and i ditched it.
i had great success with perl-firmata with several boards with V2.10 and the new V3.0 so i suggest you try it out first to determine the problem is software and not hardware or maybe OS

awong1900, 2023-04-04:
AnalogInputFirmata.cpp
[image]

I use line 92 instead of line 91. The firmata.js/example/blink.ino work fine on esp32. So i think the code line 91 not work on esp32 board. @pgrawehr

awong1900, 2023-04-04:
Like the last comment modified `AnalogInputFirmata.cpp`, The J5 test code also work well.
```
var SerialPort = require("serialport");
var five = require("johnny-five");
const port = new SerialPort("/dev/tty.wchusbserial14310", {
  baudRate: 115200,
});

var board = new five.Board({ port: port });

board.on("ready", () => {
  // Create an Led on pin 13
  const led = new five.Led(2);
  // Blink every half second
  led.blink(500);
});
```

ale-novo, 2023-04-04:
You should not be modifying the Analog input cpp
i belive J5 is not compatible with ConfigurableFirmata, but only with StandardFirmata, and that may be the reason why it was not connecting for me.

you need to flash your arduino board with this sketch, that is the one provided with the ConfigurableFirmata project.
https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino
You need to make sure you have the latest ConfigurableFirmata library installed (v3.0.0)

and then launch the j5 example, for instance
https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/blink.js

if it does not work try perl-firmata 
https://github.com/ntruchsess/perl-firmata/blob/master/examples/example-blink.pl

that is working for me for ESP8266 analog in w/o any mods

awong1900, 2023-04-06:
> You should not be modifying the Analog input cpp
> 
> i can connect successfully to ESP32 and run the capability query with perl-firmata w/o any mods and works fine.
> 
> i **believe** J5 is not compatible with ConfigurableFirmata, but only with StandardFirmata, and that may be the reason why it was not connecting for me and/or failing for you.
> 
> There is a Difference between ConfigurableFirmata and StandardFirmata. You cant run ESP32 with StandardFirmata as its not defined on boards.h
> 
> For ConfigurableFirmata: you need to flash your arduino board with this sketch, that is the one provided with the ConfigurableFirmata project. https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino You need to make sure you have the latest ConfigurableFirmata library installed (v3.0.0) and ESP32 board definition
> 
> and then launch the j5 example, for instance https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/blink.js
> 
> if it does not work try perl-firmata https://github.com/ntruchsess/perl-firmata/blob/master/examples/example-blink.pl
> 
> for me with perl-firmata and ESP32 , the capability query works fine and report analog pins, however i noticed a problem only for ESP32 where it reports initially the analog value but then freezes, ie it does not update again despite being in a loop.

OK, I'm sure you can use perl-firmata to complete the connection. You have do great test.

awong1900, 2023-04-06:
> however i noticed a problem only for ESP32 where it reports initially the analog value but then freezes, ie it does not update again despite being in a loop.

Now, it is our biggest problem.

ale-novo, 2023-04-06:
let me know if you fix it

pgrawehr, 2023-04-07:
@awong1900 Can you describe exactly what steps you're executing to freeze the ESP32? I'm not sure I'm following the scenario you have a problem with.

ale-novo, 2023-04-07:
for me i can describe what ive seen. i was thinking on creating a new issue since its not related with this 'connection' issue.

For me ESP32 analog input works, however the analog value freezes with the initial value.

lets say you connect your esp32 using firmata to a client, and query your analog pin. the pin maybe reports a value of 255 (0-1024), then if you move pot the analog pin still reports 255

other boards when you move your pot they report the right value for the new pot position.
for ESP32 there is a bug where the analog value does not get updated after the initial value is reported.

hope its clear

pgrawehr, 2023-04-08:
@ale-novo Yes, please open a separate issue on this. I noticed that I never tested analog input on ESP32, and I can reproduce that it isn't working as expected. I'm investigating.

awong1900, 2023-04-14:
I will make a new issue .

closed 2023-04-14 by awong1900