firmata/ConfigurableFirmata #141: stm32 blue pill not reporting analog pins

ale-novo, 2023-04-02:
Hello, i have a stm32 blue pill, when connecting its not reporting any analog pins however the pinout shows there are several analog pins.
my firmata sketch includes analog with `#define ENABLE_ANALOG`

```
2023-04-01 21:00:13,003 [INFO ] Connected to USB board devkit_io_serial, baudrate 115200
2023-04-01 21:00:13,607 [INFO ] ConfigurableFirmata: V_3_00, FirmataProtocol: V_2_06, SketchName: ConfigurableFirmata
2023-04-01 21:00:13,608 [INFO ] Input pins board devkit_io_serial: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
2023-04-01 21:00:13,608 [INFO ] Output pins board devkit_io_serial: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
2023-04-01 21:00:13,608 [INFO ] Analog pins board devkit_io_serial: []
2023-04-01 21:00:13,608 [INFO ] Pullup pins board devkit_io_serial: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
2023-04-01 21:00:13,608 [INFO ] Stepper pins board devkit_io_serial: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
2023-04-01 21:00:13,608 [INFO ] Encoder pins board devkit_io_serial: []
2023-04-01 21:00:13,608 [INFO ] Servo pins board devkit_io_serial: []
2023-04-01 21:00:13,608 [INFO ] Pwm pins board devkit_io_serial: [0, 1, 2, 3, 4, 5, 6, 7, 9, 12, 13, 14, 15, 20, 21, 22, 23, 26, 27, 28, 29, 30, 31]
2023-04-01 21:00:13,609 [INFO ] Firmata device connected True board devkit_io_serial
```

im looking at boards.h and seems there is consideration for analog. is there anything wrong with that configuration? cheers.

pgrawehr, 2023-04-03:
Not easy to know without looking at the board toolkit. 

I would however guess that 
```
#define TOTAL_PINS              NUM_DIGITAL_PINS
```
is probably wrong. This should probably be 
```
#define TOTAL_PINS              NUM_DIGITAL_PINS + NUM_ANALOG_INPUTS
```
for this board. But to be sure, I would need to know what the value of `A0` is and whether the digital pins include the analog ones on this board or not. Most boards can use the analog inputs also as digital pins, but maybe not this one.

ale-novo, 2023-04-04:
Hi, thanks for your response.
i also noted that the stm32 blue pill pins do not match a logical order in firmata.
The stm32 blue pill pins are separated in 3 ports, port a, port b and port c.
i had to test one by one and found these are the pins:

| firmata pin | stm32 port |
| ----------- | ---------- |
| 0 | b9 |
| 1 | b8 |
| 2 | b7 |
| 3 | b6 |
| 4 | b5 |
| 5 | b4 |
| 6 | b3 |
| 7 | a15 |
| 12 | a8 |
| 13 | b15 |
| 14 | b14 |
| 15 | b13 |
| 16 | b12 |
| 17 | c13 LED |
| 18 | c14 |
| 19 | c15 |
| 20 | a0 |
| 21 | a1 |
| 22 | a2 |
| 23 | a3 |
| 24 | a4 |
| 25 | a5 |
| 26 | a6 |
| 27 | a7 |
| 28 | b0 |
| 29 | b1 |
| 30 | b10 |
| 31 | b11 |

Analog pins should be:

Analog Pins: PA0-PA7, PB0-PB1

pgrawehr, 2023-04-07:
I can't see directly what's wrong then. The analog pins are within the digital pin range, so they should be reported correctly. 

I'm assuming you got a wrong pinout diagram there. The development boards for these chips are (unlike with Arduinos) not standardized, so different vendors could make different layouts.

ale-novo, 2023-04-07:
Hi, the table ive posted before ive constructed myself from testing one pin at a time, so its the right one
however configurable firmata is reporting that the board has no analog capable pins