firmata/ConfigurableFirmata #143: RP2040 Servo stops responding after firmata server side script killed and restart

ale-novo, 2023-04-04:
Hello, i have a Client RP2040 board with the ConfigurableFirmata v3.0 library installed on it, and an Python script using Perl-Firmata server side firmata library running on a PC via USB.
They both connect and communicate fine.

Ive noticed that with the RP2040 board, if i kill the script (control+c) and the re launch it, the board will re connect and communicate correctly, responds correctly to  capability query, etc, but the servo no longer moves, until i recycle the power on the RP2040 board.

This does not happen with for example nano and esp8266, servo responds fine after i kill the script and re launch
also this does not happen for example with I/O, stepper, encoder, analog, etc.

For some reason only RP2040 for servo is not responding after i kill the script.

i am running the firmata close method however to be completely frank im not sure if this is being executed after control+c as this is running in a subprocess for each connected board and im doing control+c from the main thread (terminal)

```
  except KeyboardInterrupt:
    logger.info("Stopping board %s", name)

    # write config when killed
    #save_config(True, accelstepper_list, stepper_list, encoder_list, button_list, led_list, switch_list, pot_list, config, config_file, timed)

    # move to init position when killed (does not work)
    #for s in stepper_list:
    #  s.start_pos()
    #for s in accelstepper_list:
    #  s.start_pos()
    logger.info("Disconnecting X-Plane")
    try:
      xp.xp_disconnect()
    except:
      pass

    logger.info("Closing firmata connection")
    firmata.close()
    logger.info("Exit")
    exit(1)
```

could the problem be failing to execute firmata close ? or just a bug with RP2040 as this does not happen in other boards?
thanks

pgrawehr, 2023-04-05:
I have never used the servo driver, but I know that it has a number of platform-specific implementations (for instance, it is completely unsupported on ESP32), so this _could_ indeed be a bug in the either the servo library, but also in the firmata code to handle it. A quick look at the code suggests that it should work with multiple initializations, but this needs further debugging.

pgrawehr, 2023-04-05:
_If_ the problem is in ConfigurableFirmata, the error is probably somewhere in `ServoFirmata.h`. I would add some logging to `ServoFiirmata.attach` to make sure the code behaves identically in both cases. Maybe it's also the analog message that doesn't get trough after the error. This command handler is a bit special as it relies on the pwm module.