firmata/ConfigurableFirmata #145: On the esp32 board, the device does not respond to the queryAnalogMapping command

awong1900, 2023-04-14:


pgrawehr, 2023-04-14:
Hmmm... I can't really confirm this, but I _can_ confirm that the reply is partially wrong (it reports an analog mapping for pins which do not even have analog support). Can you try what happens if your comment out everything but `firmataExt.reset()` from `systemResetCallback()`? I have found that the general pin resetting causes issues on ESP32, because some pins are internally reserved and shouldn't be touched.

awong1900, 2023-04-19:
> Hmmm... I can't really confirm this, but I _can_ confirm that the reply is partially wrong (it reports an analog mapping for pins which do not even have analog support). Can you try what happens if [...]

Comment like this. It's still hang up.
```
void systemResetCallback()
{
  // for (byte i = 0; i < TOTAL_PINS; i++) {
  //   if (IS_PIN_ANALOG(i)) {
  //     Firmata.setPinMode(i, PIN_MODE_ANALOG);
  //   } else if (IS_PIN_DIGITAL(i)) {
  //     Firmata.setPinMode(i, PIN_MODE_OUTPUT);
  //   }
  // }
  firmataExt.reset();
}
```

pgrawehr, 2023-04-19:
I guess you need to debug that and check the raw bytes transferred. Looking at your log above, something is wrong anyway, since the CAPABILITY_RESPONSE is much longer than what's printed above. I have no javascript-experience, so I'm unable to trace down the behavior there. Since it works with other client software, I believe it's not a firmware issue, though.

awong1900, 2023-04-20:
OK， I will check inside more.