firmata/ConfigurableFirmata #154: I need help with esp32 pins

marwans200, 2023-07-19:
pyfirmata\pyfirmata.py", line 218, in get_pin
raise InvalidPinDefError('Invalid pin definition: {0} at position 3 on {1}'
pyfirmata.pyfirmata.InvalidPinDefError: Invalid pin definition: d:16:i at position 3 on COM5

i get this error.

im a total noob trying to get input from a broken esp32 I found without wifi.
what do i do? i used pyfirmata before with an arduino uno but never with an esp32 or anything else

pgrawehr, 2023-07-23:
Unfortunately, I do not know what pyfirmata is trying to tell you there. The error message is quite unreadable. Which version of ConfigurableFirmata are you using?

Maybe @tino (maintainer of pyfirmata) has an idea what's going on. Maybe the board reports a mode that pyfirmata doesn't understand?

tino, 2023-08-21:
I'm sorry, I don't think I can help you here. The only thing I can think of is that you'll probably set up a good definition for the board, depending on the layout.

pgrawehr, 2023-09-02:
@tino I'm very poor in python, but I think the code that parses the different pin modes is to simplistic. Have you ever tested against a recent ConfigurableFirmata version? A single pin can report 10 or more different possible modes, and you seem to only expect input, output or PWM.

tino, 2023-10-03:
> Have you ever tested against a recent ConfigurableFirmata version?

Nope. If new modes are present that certainly is missing indeed.