firmata/ConfigurableFirmata #158: Fatal Error DHT.h not found

trhh07, 2023-09-22:
I have just started getting into arduino and firmata with a Nano ESP32 but every time I try and compile ConfigurebleFirmata i keep on getting this error:
**In file included from /private/var/folders/qh/_6xsmx_122n895s6__5zltmh0000gn/T/.arduinoIDE-unsaved2023822-46130-tepy15.q3pw/ConfigurableFirmata/ConfigurableFirmata.ino:93:
/Users/trhh07/Documents/Arduino/libraries/ConfigurableFirmata/src/DhtFirmata.h:26:10: fatal error: DHT.h: No such file or directory
 #include <DHT.h>**
and as I am new im not sure if there is an easy way to fix this or not. 
I am on mac if that affects anything.

pgrawehr, 2023-09-23:
You need to install the library "DHT sensor library" to fix this. Or you disable data support in the main ino file.

pgrawehr, 2023-11-07:
Closing this as "works as expected".

closed 2023-11-07 by pgrawehr

ScreamZ, 2023-11-09:
Could be nice to document this

pgrawehr, 2023-11-09:
@ScreamZ My bad, I thought it was... Hopefully, I don't forget.