firmata/ConfigurableFirmata #160: Define static IP

ScreamZ, 2023-11-09:
Hello,

I'm using this for ESP32, is there any way to define a static IP address instead of relying on DHCP ?

With standardFirmataWifi it's possible but looks like not supported for ESP32 ?
Also is there any ETA for servo ?

Or is it better that I switch to ESP8266 ?

Thank you

pgrawehr, 2023-11-10:
Hi!

> I'm using this for ESP32, is there any way to define a static IP address instead of relying on DHCP ?
I have never tried, but it should be possible by just changing a few lines in the inialization code. You might check [this example](https://randomnerdtutorials.com/esp32-static-fixed-ip-address-arduino-ide/), for instance. Instead of assigning a static IP address on the ESP32, you can usually also configure your WIFI router to assign always the same IP to this device instead. 

> Also is there any ETA for servo ?
Until very recently, the servo library was not supported on the ESP32, but I think there had been some work to fix that. So with the very latest versions of everything, it could work. I do not have any servos to test, so unfortunately you need to do the testing yourself. The first step would be to enable the servo component for the ESP32 in the ino file and check whether the build succeeds.

> Or is it better that I switch to ESP8266 ?
I don't think so, the ESP8266 is quite a step backwards in performance and capabilites from the ESP32. It does have a bit broader support, because it's older, but going forward, I would rather recommend trying to get everything to work with the newer chip.

pgrawehr, 2024-06-17:
Closing this as possibly solved

closed 2024-06-17 by pgrawehr