firmata/ConfigurableFirmata #164: Firmata USB Baud Fhem

passat92, 2024-03-16:
Hello, where do I set the baud rate in the configurable Firmata 3.2 if I want to send it to Fhem via USB?

pgrawehr, 2024-03-16:
Sorry for the maybe silly question, but what is Fhem? 
The communication baud rate is set in the main .ino file. The line reads something like `Firmata.begin(115200)`. The number is the baud rate.

passat92, 2024-03-16:
Fhem is a home automation system for a Raspberry or similar. https://de.m.wikipedia.org/wiki/FHEM Thanks for tip, now it works (was set to 56700 in fhem) so far except for i2c,  the display shows no reaction. It works with the one old Arduino mega that was programmed with Firma Builder. It doesn't work with the Firmata V 3.2

pgrawehr, 2024-03-16:
Did you enable the i2c module? 

It is possible that some changes on the host side are necessary, as there have been some changes to the i2c module. If I remember correctly, they should be backwards compatible, but obviously I couldn't test with all possible clients. Since it works with the older version, I suspect something like this.

passat92, 2024-03-16:
Yes i2c is enabled in the sketch. FHEM shows that it is aviable on pin20/21.

passat92, 2024-03-17:
> [Verfolgungsbild][https://github.com/notifications/beacon/BE4JW4NSQY4TY5QQYB6TF5LYYSV3PA5CNFSM6AAAAABEZOFJUCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTXKYDPA.gif]

Ich bekomme im FHEM im state mit der v3.2 immer wieder die Meldung
Unhandled sysex command

pgrawehr, 2024-03-17:
Das deutet schon darauf hin, dass entweder eine notwendige Komponente nicht aktiv ist oder eine Protokoll-Inkompatibilität besteht. Ändere doch mal in `FirmataExt.cpp` die Ausgabe so ab: 

```cpp
Firmata.sendStringf(F("Unhandled sysex command: 0x%x (len: %d)"), (int)command, (int)argc);
```
Dann sehen wir besser, welches Kommando hier nicht versanden wird.

passat92, 2024-03-17:
Probiere ich die Tage mal.

passat92, 2024-03-18:
Hallo Patrik,

ist das keine f hinter sensString so gewollt? Ich finde keine Zeile die so heißt. Kannst du mir eventuell die Zeilennummer sagen?

Liebe grüße, Enrico

Firmata.sendStringf(F("Unhandled sysex command: 0x%x (len: %d)"), (int)command, (int)argc);

Dann sehen wir besser, welches Kommando hier nicht verstanden wird.

pgrawehr, 2024-03-19:
Sorry, das hätte ich schreiben sollen. Ich meine die Funktion `void handleSysexCallback(byte command, byte argc, byte* argv)` in FirmataExt.cpp, Zeile 31. 

Und ja, die Methode heist `sendStringf` (Im Gegensatz zu `sendString` erlaubt diese Funktion Argumente im printf-Stil)

passat92, 2024-03-20:
Hallo,

vielen dank für deine Hilfe. Ich bin da leider nicht so firm drin. Komischer weise funktioniert es nachdem ich die Zeile hinzugefügt habe jetzt einfach von alleine. Keine Ahnung woran das jetzt lag.

Nun verzweifel ich aber am 2. Projekt und zwar möchte ich noch einen weitern 2560 mit w5100 Ethernet Shield bauen.Leider gibt es ja diese schöne Builder Seite nicht mehr bzw. nur veraltet. Nun weiß ich leider nicht was ich in der IDE machen muss damit das conf. firmata mit dem Shield spricht.

Leider gibt es ja da keine Beispiele für.

Lg Enrico

pgrawehr, 2024-03-20:
Ich habe leider keine Arduinos mit Ethernet-Shield, sondern habe dann immer gleich ESP32 verwendet. Die haben Wifi schon eingebaut und kosten trotzdem deutlich weniger. 

Du kannst versuchen, den in `ConfigurableFirmata.ino` vorhandenen WIFI-Code des ESP32 anzupassen, oder du kannst [dieses ältere Beispiel](https://github.com/firmata/ConfigurableFirmata/commit/c55e36bb0e83281a68e5b38a7620256857298a37) ansehen, vielleicht findest du dort Hinweise darauf, wie du das Ethernet-Shield einsetzen kannst. Ich hatte das Beispiel rausgeworfen, weil ich es nicht testen kann und es wahrscheinlich so inzwischen auch nicht mehr kompiliert.

passat92, 2024-03-20:
Okay, schade, ich habe ein paar von den dingern zu laufen. Ich finde die Kabelverbindung irgendwie schöner. Leider bin ich dazu zu dumm und mein Freund der mir immer mit Rat und Tat zur Seite stand hat leider Suizid begangen...  :)