firmata/ConfigurableFirmata #178: ESP32 compile issues with fix

zlewko, 2025-02-20:
I had trouble compiling for 3.3.0 for ESP32, I needed to make two changes to the library to get it to compile and work.

analogOutputFirmata.h Line 32: 
#if ESP32
changed to
#ifdef ESP32

analogOutputFirmataEsp32.cpp Line 33:
internalReset();
changed to
this->internalReset();

pgrawehr, 2025-02-21:
Thank you. That is a known issue and already fixed on master. I just didn't have time yet to complete the release. I'll try to do this soon.

PizzaProgram, 2025-04-13:
I think if we fix the SERVO part, (see [pending pull req. here](https://github.com/firmata/ConfigurableFirmata/pull/184))
Than we can normalise this part of the sample files too:
```c
// Note that the SERVO module currently is not supported on ESP32. So either disable this or patch the library
#ifndef ESP32
// #define ENABLE_SERVO 
#endif
```