firmata/ConfigurableFirmata #179: Sampling interval - only for analogue ports?

PizzaProgram, 2025-03-30:
I've just finished a complete rewrite of: [Node-Red Node Arduino (firmata)](https://discourse.nodered.org/t/announce-node-red-node-arduino-firmata-remastered-v1-3-3/96317/3), so it allows to set other Sampling intervals than default 19ms.

But I've just realised, it has a very bad side effect:
- it will reduce digital pin sampling rate too, 
- not just analogue! 
- and who knows what else? PWM? 

## is there any workaround for this?
- either setting individual pin-frequencies
- or simply define a compile constant to apply sampling rate changes to analogue ports? Like:
```
#define FAST_SAMPLING_FOR_DIGITAL
// this function will maintain sampling rate for digital ports at 10ms, 
// while analogue ports can be set differently on the go via SYSEX+ 0x7a command.
```

## Why is it important?
[Node-RED](https://nodered.org) is written in node.js, and it's running a bit slow on a tiny host-boards, like Raspberry Pi 3.
So it can not handle too much info. 10ms interval means 100 message / sec / pin. 
For example:
- it is perfectly enough to read the built-in temperature value of the RPi Pico every 1000ms to monitor danger of overheating.

Also it can reduce unnecessary traffic on the narrow 57 Kbaud bus.

---

_If somebody could drop a link to the code, where all this is happening, maybe I could come up with an idea..._

# Thanks! :-)

pgrawehr, 2025-03-30:
I can maybe take a look next week. Can you tell me what command(s) you're sending and what you would idealy expect?

PizzaProgram, 2025-04-01:
_I do not understand, why you asked me these, (except if you did not understand the request, ) but here are my (completely unrelated) answers:_

> Can you tell me what command(s) you're sending

I'm calling both:
- the standard function of firmata-io.js function, which is:

```js
const SAMPLING_INTERVAL = 0x7a;
...
  setSamplingInterval(interval) {
    const safeint = interval < 10 ? 10 : interval > 65535 ? 65535 : interval;
    this.settings.samplingInterval = safeint;
    writeToTransport(this, [
      START_SYSEX,
      SAMPLING_INTERVAL,
      safeint & 0x7f,
      (safeint >> 7) & 0x7f,
      END_SYSEX,
    ]);
  }
```
- and at initialization changing the open parameters:
```js
const defaults = {
  reportVersionTimeout: 5000,
  samplingInterval: 19,  // << changing this too
  serialport: {
    baudRate: 57600,
    highWaterMark: 256,
    path: port,
  },
};
```

The new setting "succeeding", by both analogue + digital sampling rate changes.

>  and what you would ideally expect?

wrote it down in my 1th (issue opening) message. Please scroll up!
Any solution is fine, I can adapt.

---

### IMHO Not the "calling" part matters, but how to fix it on the Firmata's **firmware** side.
Of course it would be awesome, if it would be also possible to change it runtime, via calling, not just at `void setup()` on the firmware side! _(If that's why you have asked these ?)_

pgrawehr, 2025-04-01:
I asked the question because you might have a particular expectation on what feature you expect in detail. Do you want that the command just updates the analog reporting frequency, or should I extend the protocol so that one can specify (with some limits) the update rate for e.g. analog and digital separately?

PizzaProgram, 2025-04-01:
> I asked the question because you might have a particular expectation on what feature you expect in detail.

OK, sorry! And Thank you ! :-)

> Do you want that the command just updates the analog reporting frequency, or should I extend the protocol so that one can specify (with some limits) the update rate for e.g. analog and digital separately?

Both solutions are fine!
Whatever is easier to do for you to do. ;-)

### Basically: 
- setting a global value + changing analogue only, 
  _has same effect, as:_
- setting both separately 

Of course the most-most perfect solution would be to set each pin individually, but I would not dream about it.
**Again:** if I would know which part of the code is responsible for doing the :
- normal loop,
- frequency (PWM ?)
- sampling rate,

than maybe I could understand how it works general + come up with (plan) some easy solution.

PizzaProgram, 2025-04-04:
_... just found an interesting fact: maybe we're just banging on an open gate:_

If [interrupt for RPi-Pico's digital pins](https://github.com/firmata/ConfigurableFirmata/issues/133) would work, this whole adjusting would be unnecessary.

pgrawehr, 2025-04-04:
I think I found the bug. In `DigitalInputFirmata::report(bool elapsed)` it checks for the "elapsed" condition, which is true exactly when the analog reporting interval has elapsed. If this condition is removed, the analog report frequency is independent of the digital report frequency. Just make sure digital input reporting is not enabled for any pins that are floating.

PizzaProgram, 2025-04-06:
@pgrawehr  Excellent news! Great job. 

Shared the info on [NR's forum too.](https://discourse.nodered.org/t/announce-node-red-node-arduino-firmata-remastered-v1-3-4-20250331/96317/6?u=pizzaprogram)

closed 2025-04-07 by pgrawehr

PizzaProgram, 2025-04-14:
on more good news:

## Tested: PWM is not affected by changing sample rate. :-)

So it is safe to lower analogue read rate by setting a high SamplingRate.

PR #181 (2025-04-07): Digital inputs should be reported independent of the sampling interval
Fixes #179